anqixu/ueye_cam #17: Unable to launch ueye XS in ros indigo, API 4.61

mklingen, 2015-05-15:
Simply trying to launch the `xs_bayer_rggb.launch` file with my ueye XS camera I get this error:

` ``
[ INFO] [1431717546.571885057]: Unable to open camera calibration file [/home/mklingen/.ros/camera_info/camera.yaml]
[ WARN] [1431717558.304534102]: Internal image scaling is not supported by camera
[ WARN] [1431717558.305973604]: Auto frame rate mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ WARN] [1431717558.306020556]: Auto exposure mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ WARN] [1431717558.627241852]: Failed to set white balance red/blue offsets to 0 / 0 for UEye camera 'camera'
[ERROR] [1431717558.627325580]: Failed to initialize UEye camera 'camera'
```

anqixu, 2015-05-18:
Try code again using rev e164973de7d30f612ef395301ddc6250737d5313.

Your previously posted error messages seem to suggest that the XS camera being used has a limited feature set. Since I don't have an XS camera on hand, we will need to debug this issue together.

mklingen, 2015-05-18:
thanks for the rapid response, here are the errors:

```
    [ WARN] [1431962044.713433898]: Internal image scaling is not supported by camera
[ WARN] [1431962044.714947984]: Auto frame rate mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ WARN] [1431962044.715011831]: Auto exposure mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ WARN] [1431962045.036415702]: Failed to set white balance red/blue offsets to 0 / 0 for UEye camera 'camera' (IS_ACCESS_VIOLATION)
[ERROR] [1431962045.036502901]: Failed to initialize UEye camera 'camera'
```
This is the exact camera I have:

https://en.ids-imaging.com/xs-home.html

anqixu, 2015-05-18:
I am perplexed as to where the code is failing. The last error message in your log suggests that the function UEyeCamDriver::connectCam() is failing. Every return statement in that function, except for the final one, is accompanied with an ERROR_STREAM() print-out. Weirdly, I don't any such error printouts in your posted log.

Could you try identifying which "is_*()" function call is failing inside connectCam(), e.g. using extra print statements or a debugger?

mklingen, 2015-05-18:
I added additional print statements. Connectcam succeeds, but reading the ROS parameter file fails:

```
Entering connectCam
ConnectCam succeeded!
[ WARN] [1431963349.545468782]: Internal image scaling is not supported by camera
[ WARN] [1431963349.547149458]: Auto frame rate mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ WARN] [1431963349.547229817]: Auto exposure mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ WARN] [1431963349.893595235]: Failed to set white balance red/blue offsets to 0 / 0 for UEye camera 'camera' (IS_ACCESS_VIOLATION)
ROS param failed with IS_ACCESS_VIOLATION
[ERROR] [1431963349.893932950]: Failed to initialize UEye camera 'camera'

```

mklingen, 2015-05-18:
simply commenting out the return there (at parsing the ROS param file) so that it returns `IS_SUCCESS` gets it to work, and it now spits out images. The images appear to be interlaced mono images (I'm guessing raw bayer images)

mklingen, 2015-05-18:
I can now get color RGB images using the debayer, however I did have to comment out the check for the access violation on setting the white balance to get it to work.

anqixu, 2015-05-18:
Please confirm that bf8b57c8eaaa253a435371c4e18d52ecb53e74cc works with your setup.

mklingen, 2015-05-18:
Works, thanks!

closed 2015-05-18 by anqixu