anqixu/ueye_cam #18: XS camera not working at full resolution

phil333, 2015-06-11:
I have been able to use the XS camera at a resolution of 1280x720.

Once I try to change to resolution with the rqt_reconfigure or in the launch file, (also if i have a correct .ini file) i get the following error:

Failed to set UEye camera sensor's Area Of Interest to 2560 x 1920 with top-left corner at (0, 0)

Any suggestions?

anqixu, 2015-06-15:
According to the specs, uEye XS's maximum resolution is 2592 x 1944. I won't be surprised if 2560x1920 resolution is only compatible with low frame rate, high clock rate, and other matching settings.

I recommend to configure your uEye XS camera in ueyedemo, obtain the settings for ALL parameters that ueye_cam supports (see default launch/rgb8.launch), and configure them accordingly.

If that still doesn't work, please provide the uEye camera configuration ini file, the ROS launch file, and any other relevant information. Note that I don't have an uEye XS camera, so we will have to look into this further, together.

phil333, 2015-06-16:
Hey, I ve tried different options, but it doesn't want to change the settings, not from the start, now on the fly (with rqt_reconfigure)

this is the launch file I made:
https://www.dropbox.com/s/uqmx0zl00r5px40/rgb8_XS.launch?dl=0
and the .ini file coming directly from the ids cameramanager
https://www.dropbox.com/s/y1wyawtlo80lgb3/XS_conf.ini?dl=0

[ WARN] [1434471074.586388716]: Current camera color mode is not supported by this wrapper;supported modes: {MONO8 | RGB8 | BAYER_RGGB8}; switching to RGB8 (24-bit)
[ERROR] [1434471074.586454830]: Could not set color mode to RGB8 (IS_INVALID_COLOR_FORMAT)
[ WARN] [1434471074.586490930]: Camera configuration loaded into an unsupported color mode; switching to MONO8.
[ERROR] [1434471079.957305317]: Failed to set UEye camera sensor's Area Of Interest to 2560 x 1920 with top-left corner at (16, 12)
[ERROR] [1434471079.957355847]: Failed to initialize UEye camera 'camera'

When i use the XS launch file that is provided, everything works fine as long as i don't try to change the resolution. I only get the following warnings at the start:

[ INFO] [1434471280.671458382]: Initializing nodelet with 4 worker threads.
process[ueye_cam_nodelet-2]: started with pid [2553]
[ INFO] [1434471280.765249382]: Unable to open camera calibration file [/home/user/.ros/camera_info/camera.yaml]
[ WARN] [1434471313.002446728]: Internal image scaling is not supported by camera
[ WARN] [1434471313.003866404]: Auto frame rate mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ WARN] [1434471313.003923616]: Auto exposure mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ WARN] [1434471313.347196003]: Failed to set white balance red/blue offsets to 0 / 0 for UEye camera 'camera' (IS_ACCESS_VIOLATION)
[ WARN] [1434471314.763515260]: Auto frame rate mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ WARN] [1434471314.763596319]: Auto exposure mode is not supported for UEye camera 'camera' (IS_NOT_SUPPORTED)
[ INFO] [1434471314.782726720]: UEye camera 'camera' initialized on topic /camera/image_raw
Width:                  1280
Height:                 720
Left Pos.:              0
Top Pos.:               0
Color Mode:             bayer_rggb8
Subsampling:            1
Binning:                1
Sensor Scaling:         1
Auto Gain:              0
Master Gain:            0
Red Gain:               0
Green Gain:             1
Blue Gain:              16
Gain Boost:             0
Auto Exposure:          0
Exposure (ms):          33
Auto White Balance:     1
WB Red Offset:          0
WB Blue Offset:         0
Flash Delay (us):       0
Flash Duration (us):    1000
Ext Trigger Mode:       0
Auto Frame Rate:        0
Frame Rate (Hz):        11.6307
Pixel Clock (MHz):      30
Mirror Image Upside Down:       0
Mirror Image Left Right:        0

anqixu, 2015-06-16:
I found a number of discrepancies between your launch and ini files. Please carefully go through them and make sure ALL PARAMETERS match (see below for details).

Also, first start ueyedemo and make sure all settings match the ini file (easiest way would be to load ini file). Then, save the settings into the camera's EEPROM. Next, start the ROS launch file. You should no longer see warnings like "current camera color mode is not supported..." from the ROS launch.

Here are some parameters that are incorrectly configured in the launch file:

-> image_left/image_top
Start X=0
Start Y=0

Pixelclock=30
Framerate=2.787047
Exposure=33.227333

-> gain params
Master=0
Red=0
Green=0
Blue=0
GainBoost=7

-> ueye_cam doesn't support this... see if you can disable this in ueyedemo
JpegCompression=2

-> also turn off auto exposure

phil333, 2015-07-08:
Thank you for the tips. I have tried even with multiple launchfile changes and i can't get it to work. How can i save settings into the camera's EEPROM? I suspect it might be a color mode problem. Do you have any idea how i can permenently set the colour mode as well as switching of the Jpeg mode?

anqixu, 2015-08-03:
- run ueyedemo, open your camera (live), open the Properties popup, change the color mode settings (must use DIB mode)

- I'm not sure how to switch out of JPEG compression mode, or whether your camera even supports non-compressed modes. You will need to look for it in the Properties popup

- in the Properties popup, one of the tabs (I think it's the last one) has buttons that allow you to write your settings to the camera's EEPROM; alternatively, in File menu, there should be a "save parameters..." option that you can use

- pull revision 19bba44a5d6939f08e5fb96f8648495ed453c672

- either turn on ROS debug messages when running nodelet, or modify the following in include/ueye_cam/logging_macros.hpp:

  ```
  line 84:   #define DEBUG(...) NODELET_INFO(__VA_ARGS__)
  line 89:   #define DEBUG_STREAM(...) NODELET_INFO_STREAM(__VA_ARGS__)
  ```

- compile ueye_cam from scratch

- run your launch file, and report back with all of its printouts

anqixu, 2015-08-06:
The latest revision of the codebase also explicitly checks if camera is set to JPEG mode, and attempts to set it back to MONO8 mode. That should solve your problem.

closed 2015-08-06 by anqixu