anqixu/ueye_cam #19: [reconfig GUI] Problem with external trigger setting after changing the exposure time

Chris-Bee, 2015-06-19:
I use the external trigger function with a UI-3240LE. Everything is fine until I change the exposure time in the reconfig gui. The check box for the external trigger is still active but its not working. In this situation, if I turn the external trigger in the reconfig gui off and on, its working from that point unless I change the exposure time again.

Thanks in advance.

Chris

anqixu, 2015-06-26:
I can't think of a clear reason why this problem is occurring. Unfortunately I don't have access to any uEye cameras right now, so we will need to work together to find out more about this issue.

--- (A) ---

I presume that the following steps will recreate the issue:
1. launch 2 ueye_cam nodelets for master (free-run mode) and slave (external trigger mode) cameras
2. subscribe to both /master/image_raw and /slave/image_raw topics (otherwise master will not send trigger pulses and slave will be in idle mode); observe that slave camera is capturing frames
3. in rqt_reconfigure, change the slave nodelet's 'exposure' parameter
4. observe that slave camera is no longer capturing frames

If the above is not correct, then indicate your sequence, and also try the above sequence of events and report back.

--- (B) ---

If possible, check to see that the master camera continues to capture frames, and continues to emit active-low signals on the trigger output pin (using an oscilloscope).

--- (C) ---

After looking at the ueye_cam_nodelet codebase, I've determined that changing exposure settings should not affect the camera's free-run / external trigger mode settings. However, due to a recent change to the codebase, the nodelet now ignores any errors returned by the setExposure() function. You should try disabling this feature, by making the following change to ueye_cam/src/ueye_cam_nodelet.cpp, line 487, from:

if ((is_err = setExposure(cam_params_.auto_exposure, cam_params_.exposure)) != IS_SUCCESS) noop;

to

if ((is_err = setExposure(cam_params_.auto_exposure, cam_params_.exposure)) != IS_SUCCESS) return is_err;

This is only a precaution, and I don't expect it to identify the culprit, since the setExposure() function itself will print warning or error messages if things went wrong.

You can also try turning on the DEBUG level messages for the nodelet, and see if any of those give you hints as to what the problem is, e.g. did the exposure setting change to the desired value?

--- (D) ---

Note that the settings shown by dynamic_reconfigure reflect the state of the ueye_cam ROS driver wrapper, which may or may not reflect the actual state of the underlying IDS driver. Thus, to debug further, I would suggest querying and printing the exposure and external trigger mode states of the camera at key moments.

One way this might be done is to add some code inside the "while (frame_grab_alive_ && ros::ok())" loop in UEyeCamNodelet::frameGrabLoop(), e.g.:

double curr_exposure_ms = -1;
INT is_err_exposure = is_Exposure(cam_handle_, IS_EXPOSURE_CMD_SET_EXPOSURE,
        (void*) &(curr_exposure_ms ), sizeof(curr_exposure_ms ));

INT curr_trigger_mode = is_SetExternalTrigger (cam_handle_, IS_GET_EXTERNALTRIGGER);

NODELET_INFO_STREAM(cam_name_  << " curr_exposure_ms : " << curr_exposure_ms << ", curr_trigger_mode: " << curr_trigger_mode << ", external trigger on? " << (curr_trigger_mode == IS_SET_TRIGGER_HI_LO));
// curr_exposure_ms values can also be compared against:
// IS_SET_TRIGGER_OFF
// IS_SET_TRIGGER_HI_LO
// IS_SET_TRIGGER_LO_HI
// IS_SET_TRIGGER_PRE_HI_LO
// IS_SET_TRIGGER_PRE_LO_HI
// IS_SET_TRIGGER_HI_LO_SYNC
// IS_SET_TRIGGER_LO_HI_SYNC

anqixu, 2015-08-03:
I'm closing this issue for now due to inactivity. Please report back if you are continuing to experience woes. Also:

- pull revision 19bba44a5d6939f08e5fb96f8648495ed453c672 (or later)

- either turn on ROS debug messages when running nodelet, or modify the following in include/ueye_cam/logging_macros.hpp:

  ```
  line 84:   #define DEBUG(...) NODELET_INFO(__VA_ARGS__)
  line 89:   #define DEBUG_STREAM(...) NODELET_INFO_STREAM(__VA_ARGS__)
  ```

- compile ueye_cam from scratch

- run your launch file, and report back with all of its printouts

closed 2015-08-03 by anqixu