anqixu/ueye_cam #20: UI-1220LE-M-GL color_mode woes

jmtatsch, 2015-07-24:
Hi,

i know the UI-1220LE-M-GL is not officially supported, but it seems there is just a minor fix required to get it running properly. When the nodulet is started from a launch file with mono8 and 640x480 it will produce the below log and does nothing. When dynamic reconfigure is used to set color_mode again to mono8 it works like a charm. I will post the full launch file as soon as I have access to it again. Am I using it wrong or what could be the issue here?

```
roslaunch ueye_cam bw.launch
... logging to /home/odroid/.ros/log/b9ce0c42-315c-11e5-a17f-001e06617a39/roslaunch-odroid-10559.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://odroid:41110/

SUMMARY
========

PARAMETERS
 * /rosdistro: jade
 * /rosversion: 1.11.13
 * /ueye_cam_nodelet/auto_exposure: True
 * /ueye_cam_nodelet/auto_frame_rate: False
 * /ueye_cam_nodelet/auto_gain: True
 * /ueye_cam_nodelet/auto_white_balance: True
 * /ueye_cam_nodelet/binning: 1
 * /ueye_cam_nodelet/blue_gain: 16
 * /ueye_cam_nodelet/camera_id: 0
 * /ueye_cam_nodelet/camera_intrinsics_file: 
 * /ueye_cam_nodelet/camera_name: camera
 * /ueye_cam_nodelet/camera_parameters_file: 
 * /ueye_cam_nodelet/camera_topic: image_raw
 * /ueye_cam_nodelet/color_mode: mono8
 * /ueye_cam_nodelet/exposure: 33
 * /ueye_cam_nodelet/ext_trigger_mode: False
 * /ueye_cam_nodelet/flash_delay: 0
 * /ueye_cam_nodelet/flash_duration: 1000
 * /ueye_cam_nodelet/flip_lr: False
 * /ueye_cam_nodelet/flip_upd: False
 * /ueye_cam_nodelet/frame_rate: 30.0
 * /ueye_cam_nodelet/gain_boost: False
 * /ueye_cam_nodelet/green_gain: 1
 * /ueye_cam_nodelet/image_height: 480
 * /ueye_cam_nodelet/image_left: -1
 * /ueye_cam_nodelet/image_top: -1
 * /ueye_cam_nodelet/image_width: 640
 * /ueye_cam_nodelet/master_gain: 0
 * /ueye_cam_nodelet/pixel_clock: 25
 * /ueye_cam_nodelet/red_gain: 0
 * /ueye_cam_nodelet/sensor_scaling: 1.0
 * /ueye_cam_nodelet/subsampling: 1
 * /ueye_cam_nodelet/white_balance_blue_offset: 0
 * /ueye_cam_nodelet/white_balance_red_offset: 0

NODES
  /
    nodelet_manager (nodelet/nodelet)
    ueye_cam_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [10570]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b9ce0c42-315c-11e5-a17f-001e06617a39
process[rosout-1]: started with pid [10583]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [10594]
process[ueye_cam_nodelet-3]: started with pid [10601]
[ INFO] [1437671004.974484114]: Initializing nodelet with 8 worker threads.
[ INFO] [1437671005.658660640]: Unable to open camera calibration file [/home/odroid/.ros/camera_info/camera.yaml]
[ WARN] [1437671012.177399759]: Could not load UEye camera 'camera' sensor parameters file /home/odroid/.ros/camera_conf/camera.ini (IS_FILE_WRITE_OPEN_ERROR)
[ WARN] [1437671012.180301971]: Camera configuration loaded into an unsupported color mode; switching to MONO8.
[ WARN] [1437671013.117394007]: Internal image scaling is not supported by camera
[ INFO] [1437671013.739942835]: UEye camera 'camera' initialized on topic /camera/image_raw
Width:			640
Height:			480
Left Pos.:		-1
Top Pos.:		-1
Color Mode:		mono8
Subsampling:		1
Binning:		1
Sensor Scaling:		1
Auto Gain:		1
Master Gain:		0
Red Gain:		0
Green Gain:		1
Blue Gain:		16
Gain Boost:		0
Auto Exposure:		1
Exposure (ms):		33
Auto White Balance:	1
WB Red Offset:		0
WB Blue Offset:		0
Flash Delay (us):	0
Flash Duration (us):	1000
Ext Trigger Mode:	0
Auto Frame Rate:	0
Frame Rate (Hz):	30.0059
Pixel Clock (MHz):	25
Mirror Image Upside Down:	0
Mirror Image Left Right:	0

[ INFO] [1437671089.123576095]: Camera camera set to free-run mode
[ERROR] [1437671089.125415180]: Camera buffer pitch (752) is smaller than expected: width (640) * bytes per pixel (1)
[ERROR] [1437671089.132365397]: Camera buffer pitch (752) is smaller than expected: width (640) * bytes per pixel (1)
[ERROR] [1437671089.146388956]: Camera buffer pitch (752) is smaller than expected: width (640) * bytes per pixel (1)
[ERROR] [1437671089.179405081]: Camera buffer pitch (752) is smaller than expected: width (640) * bytes per pixel (1)
[ERROR] [1437671089.213082623]: Camera buffer pitch (752) is smaller than expected: width (640) * bytes per pixel (1)
```

anqixu, 2015-07-24:
I think you've helped me find 2 potential bugs, which is great! Please try out each of the following section, re-compile, and report back.

----- 1 ------

in src/ueye_cam_nodelet.cpp, edit line 1021 to the following:

        if (expected_row_stride > cam_buffer_pitch_) { // was previously <

----- 2 ------

First undo the previous change. Then, in include/ueye_cam/ueye_cam_nodelet.hpp, edit lines 77 and 78 to the following:

  constexpr static int DEFAULT_IMAGE_WIDTH = -1; // was previously 640
  constexpr static int DEFAULT_IMAGE_HEIGHT = -1; // was previously 480

jmtatsch, 2015-07-30:
Fix 1
as soon as I subscribe to the topic 

'[ERROR] [1438270744.710797803]: Camera buffer pitch (752) is smaller than expected: width (640) * bytes per pixel (1)'

jmtatsch, 2015-07-30:
No 2

```
roslaunch ueye_cam bw.launch 
... logging to /home/odroid/.ros/log/a8841604-36d2-11e5-a38c-001e06617a39/roslaunch-odroid-27843.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://odroid:55917/

SUMMARY
========

PARAMETERS
 * /rosdistro: jade
 * /rosversion: 1.11.13
 * /ueye_cam_nodelet/auto_exposure: True
 * /ueye_cam_nodelet/auto_frame_rate: False
 * /ueye_cam_nodelet/auto_gain: True
 * /ueye_cam_nodelet/auto_white_balance: True
 * /ueye_cam_nodelet/binning: 1
 * /ueye_cam_nodelet/blue_gain: 16
 * /ueye_cam_nodelet/camera_id: 0
 * /ueye_cam_nodelet/camera_intrinsics_file: 
 * /ueye_cam_nodelet/camera_name: camera
 * /ueye_cam_nodelet/camera_parameters_file: 
 * /ueye_cam_nodelet/camera_topic: image_raw
 * /ueye_cam_nodelet/color_mode: mono8
 * /ueye_cam_nodelet/exposure: 33
 * /ueye_cam_nodelet/ext_trigger_mode: False
 * /ueye_cam_nodelet/flash_delay: 0
 * /ueye_cam_nodelet/flash_duration: 1000
 * /ueye_cam_nodelet/flip_lr: False
 * /ueye_cam_nodelet/flip_upd: False
 * /ueye_cam_nodelet/frame_rate: 30.0
 * /ueye_cam_nodelet/gain_boost: False
 * /ueye_cam_nodelet/green_gain: 1
 * /ueye_cam_nodelet/image_height: 480
 * /ueye_cam_nodelet/image_left: -1
 * /ueye_cam_nodelet/image_top: -1
 * /ueye_cam_nodelet/image_width: 640
 * /ueye_cam_nodelet/master_gain: 0
 * /ueye_cam_nodelet/pixel_clock: 25
 * /ueye_cam_nodelet/red_gain: 0
 * /ueye_cam_nodelet/subsampling: 1
 * /ueye_cam_nodelet/white_balance_blue_offset: 0
 * /ueye_cam_nodelet/white_balance_red_offset: 0

NODES
  /
    nodelet_manager (nodelet/nodelet)
    ueye_cam_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [27854]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a8841604-36d2-11e5-a38c-001e06617a39
process[rosout-1]: started with pid [27867]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [27884]
[ INFO] [1438271412.052242194]: Initializing nodelet with 8 worker threads.
process[ueye_cam_nodelet-3]: started with pid [27906]
[ INFO] [1438271412.445352548]: Unable to open camera calibration file [/home/odroid/.ros/camera_info/camera.yaml]
[ WARN] [1438271418.724791402]: Could not load UEye camera 'camera' sensor parameters file /home/odroid/.ros/camera_conf/camera.ini (IS_FILE_WRITE_OPEN_ERROR)
[ WARN] [1438271418.727979700]: Camera configuration loaded into an unsupported color mode; switching to MONO8.
[ WARN] [1438271419.431174036]: Internal image scaling is not supported by camera
[ INFO] [1438271420.028422581]: UEye camera 'camera' initialized on topic /camera/image_raw
Width:			640
Height:			480
Left Pos.:		-1
Top Pos.:		-1
Color Mode:		mono8
Subsampling:		1
Binning:		1
Sensor Scaling:		1
Auto Gain:		1
Master Gain:		0
Red Gain:		0
Green Gain:		1
Blue Gain:		16
Gain Boost:		0
Auto Exposure:		1
Exposure (ms):		33
Auto White Balance:	1
WB Red Offset:		0
WB Blue Offset:		0
Flash Delay (us):	0
Flash Duration (us):	1000
Ext Trigger Mode:	0
Auto Frame Rate:	0
Frame Rate (Hz):	30.0059
Pixel Clock (MHz):	25
Mirror Image Upside Down:	0
Mirror Image Left Right:	0
[ INFO] [1438271434.879599300]: Camera camera set to free-run mode
[ERROR] [1438271434.881318303]: Camera buffer pitch (752) is smaller than expected: width (640) * bytes per pixel (1)
[ERROR] [1438271434.887045689]: Camera buffer pitch (752) is smaller than expected: width (640) * bytes per pixel (1)
```

anqixu, 2015-07-30:
I'm quite perplexed by your latest reports, as fix 1 appears has not been applied, given that the error message contradicts what the fix was supposed to do. (see for yourself -- lines 1021-1023 of ueye_cam_nodelet.cpp)

The following are some straight-forward and obvious suggestions:
- make sure you have uninstalled any ros-[distro]-ueye-cam from apt
- if you have multiple catkin overlays, make sure to do the modification on the copy of ueye_cam source (assuming you have multiple) at the highest overlay
- for safety, I would recommend only having one copy of ueye_cam source
- to ensure that you are indeed working on that copy, I would delete all compiled entries (naively removing build/devel/install folders from the catkin overlay, or selectively remove ueye_cam-related entries from those folders). This way when you try to roslaunch, it will fail to find any executables and you'll know that there aren't other copies of ueye_cam being selected
- make sure to git pull the latest revision before applying any fixes
- when compiling, ideally compile from scratch
- explicitly verify that compilation was 100% successful

jmtatsch, 2015-07-30:
I'm sorry, I ran with `catkin_make clean` this time and the said errors are gone. However, ueye-cam still ignores my choice of mono8 in the launch file.

anqixu, 2015-08-03:
1. please pull revision 19bba44a5d6939f08e5fb96f8648495ed453c672

2. either turn on ROS debug messages when launching node, or modify the following in include/ueye_cam/logging_macros.hpp:

line 84:   #define DEBUG(...) NODELET_INFO(__VA_ARGS__)
line 89:   #define DEBUG_STREAM(...) NODELET_INFO_STREAM(__VA_ARGS__)

3. compile ueye_cam from scratch

4. run your launch file, and report back with all of its printouts

5. please describe what "ignores my choice of mono8" means, and the steps you took to resolve them. Ideally, when pasting the printouts here, indicate when you made each change

jmtatsch, 2015-08-05:
```
roslaunch ueye_cam  bw.launch
... logging to /home/odroid/.ros/log/51959438-3b8e-11e5-9610-001e06617a39/roslaunch-odroid-31214.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://odroid:39030/

SUMMARY
========

PARAMETERS
 * /rosdistro: jade
 * /rosversion: 1.11.13
 * /ueye_cam_nodelet/auto_exposure: True
 * /ueye_cam_nodelet/auto_frame_rate: False
 * /ueye_cam_nodelet/auto_gain: True
 * /ueye_cam_nodelet/auto_white_balance: True
 * /ueye_cam_nodelet/binning: 1
 * /ueye_cam_nodelet/blue_gain: 16
 * /ueye_cam_nodelet/camera_id: 0
 * /ueye_cam_nodelet/camera_intrinsics_file: 
 * /ueye_cam_nodelet/camera_name: camera
 * /ueye_cam_nodelet/camera_parameters_file: 
 * /ueye_cam_nodelet/camera_topic: image_raw
 * /ueye_cam_nodelet/color_mode: mono8
 * /ueye_cam_nodelet/exposure: 33
 * /ueye_cam_nodelet/ext_trigger_mode: False
 * /ueye_cam_nodelet/flash_delay: 0
 * /ueye_cam_nodelet/flash_duration: 1000
 * /ueye_cam_nodelet/flip_lr: False
 * /ueye_cam_nodelet/flip_upd: False
 * /ueye_cam_nodelet/frame_rate: 40.0
 * /ueye_cam_nodelet/gain_boost: False
 * /ueye_cam_nodelet/green_gain: 1
 * /ueye_cam_nodelet/image_height: 480
 * /ueye_cam_nodelet/image_left: -1
 * /ueye_cam_nodelet/image_top: -1
 * /ueye_cam_nodelet/image_width: 640
 * /ueye_cam_nodelet/master_gain: 0
 * /ueye_cam_nodelet/pixel_clock: 25
 * /ueye_cam_nodelet/red_gain: 0
 * /ueye_cam_nodelet/subsampling: 1
 * /ueye_cam_nodelet/white_balance_blue_offset: 0
 * /ueye_cam_nodelet/white_balance_red_offset: 0

NODES
  /
    nodelet_manager (nodelet/nodelet)
    ueye_cam_nodelet (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[nodelet_manager-1]: started with pid [31232]
[ INFO] [1438796237.321315264]: Initializing nodelet with 8 worker threads.
process[ueye_cam_nodelet-2]: started with pid [31256]
[ INFO] [1438796237.816535876]: Unable to open camera calibration file [/home/odroid/.ros/camera_info/camera.yaml]
[ INFO] [1438796243.933217764]: Allocate internal memory - width: 640; height: 480; fetched pitch: 640; expected bpp: 8; total size: 307200
[ INFO] [1438796243.935687936]: Connected to UEye camera 'camera'
[ WARN] [1438796243.943328869]: Could not load UEye camera 'camera' sensor parameters file /home/odroid/.ros/camera_conf/camera.ini (IS_FILE_WRITE_OPEN_ERROR)
[ WARN] [1438796243.944319413]: Camera configuration loaded into an unsupported color mode; switching to MONO8.
[ INFO] [1438796243.946793876]: Updated color mode to mono8
[ INFO] [1438796243.950721468]: Allocate internal memory - width: 640; height: 480; fetched pitch: 640; expected bpp: 8; total size: 307200
[ INFO] [1438796243.954786809]: Allocated space for ROS image buffer:
  size = 307200
  width = 752
  height = 480
  step = 640
  encoding = mono8
[ INFO] [1438796243.956010645]: Successfully queries parameters from UEye camera 'camera'
[ INFO] [1438796244.433070346]: Updated color mode to mono8
[ INFO] [1438796244.441387947]: Allocate internal memory - width: 752; height: 480; fetched pitch: 752; expected bpp: 8; total size: 360960
[ INFO] [1438796244.603841660]: Updated resolution to 640 x 480 @ (-1, -1)
[ INFO] [1438796244.606481624]: Updated subsampling rate to 1X
[ INFO] [1438796244.608317836]: Updated binning rate to 1X
[ WARN] [1438796244.610593049]: Internal image scaling is not supported by camera
[ INFO] [1438796244.622547324]: (Re-)allocated space for ROS image buffer:
  size = 360960
  width = 640
  height = 480
  step = 752
  encoding = mono8
[ INFO] [1438796244.632765637]: Updated gain: auto
[ INFO] [1438796244.677742314]: Updated pixel clock: 25 MHz
[ INFO] [1438796244.701333946]: Updated frame rate: 40.044081 Hz
[ INFO] [1438796244.719204067]: Updated exposure: auto ms
[ INFO] [1438796244.721795072]: Updated white balance: auto
   - red offset: 0
   - blue offset: 0
[ INFO] [1438796244.737902023]: Successfully applied settings from ROS params to UEye camera 'camera'
[ INFO] [1438796244.768022127]: Updated frame rate: 40.044081 Hz
[ INFO] [1438796244.769861797]: Successfully applied settings from dyncfg to UEye camera 'camera'
[ INFO] [1438796245.240884403]: Starting threaded frame grabber loop for UEye camera 'camera'
[ INFO] [1438796245.242684990]: UEye camera 'camera' initialized on topic /camera/image_raw
Width:			640
Height:			480
Left Pos.:		-1
Top Pos.:		-1
Color Mode:		mono8
Subsampling:		1
Binning:		1
Sensor Scaling:		1
Auto Gain:		1
Master Gain:		0
Red Gain:		0
Green Gain:		1
Blue Gain:		16
Gain Boost:		0
Auto Exposure:		1
Exposure (ms):		33
Auto White Balance:	1
WB Red Offset:		0
WB Blue Offset:		0
Flash Delay (us):	0
Flash Duration (us):	1000
Ext Trigger Mode:	0
Auto Frame Rate:	0
Frame Rate (Hz):	40.0441
Pixel Clock (MHz):	25
Mirror Image Upside Down:	0
Mirror Image Left Right:	0

[ INFO] [1438796248.394099511]: Started live video mode on UEye camera 'camera'
[ INFO] [1438796248.395756389]: UEye camera 'camera' set to free-run mode
[ERROR] [1438796248.476317974]: Timed out while acquiring image from UEye camera 'camera' (IS_TIMED_OUT)
[ERROR] [1438796248.476876016]: If this is occurring frequently, see https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549
[ERROR] [1438796248.562016194]: Timed out while acquiring image from UEye camera 'camera' (IS_TIMED_OUT)
```

"ignores my choice of mono8" means that when starting from launch file, it ignores that the color mode was set to mono8 there. If the color mode is set later by dynamic reconfigure, it is handled as expected.

anqixu, 2015-08-05:
First, I want to thank you for helping me discover some bugs when initializing the camera. I'm not sure if these are the cause of your woes, so please pull the latest revision (564152225bd3a9fbc8b24fdc7ad386fa0a263f73 or later), ensure that DEBUG_STREAM is still INFO_STREAM, and try your launch file again.

Also, after roslaunch-ing, please report the output of 'rostopic echo <camera_image_topic>' (i.e. I'm curious whether the nodelet publishes non-sensical data, or if it does not publish any data at all). Furthermore, after you go in dyncfg to toggle the color mode to MONO8, paste whatever is printed by the nodelet here.

jmtatsch, 2015-08-06:
Yupp, it now seems to handle the color mode correctly.

```
roslaunch ueye_cam bw.launch
... logging to /home/odroid/.ros/log/dc44ff70-3c41-11e5-b40e-001e06617a39/roslaunch-odroid-2830.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://odroid:55088/

SUMMARY
========

PARAMETERS
 * /rosdistro: jade
 * /rosversion: 1.11.13
 * /ueye_cam_nodelet/auto_exposure: True
 * /ueye_cam_nodelet/auto_frame_rate: False
 * /ueye_cam_nodelet/auto_gain: True
 * /ueye_cam_nodelet/auto_white_balance: True
 * /ueye_cam_nodelet/binning: 1
 * /ueye_cam_nodelet/blue_gain: 16
 * /ueye_cam_nodelet/camera_id: 0
 * /ueye_cam_nodelet/camera_intrinsics_file: 
 * /ueye_cam_nodelet/camera_name: camera
 * /ueye_cam_nodelet/camera_parameters_file: 
 * /ueye_cam_nodelet/camera_topic: image_raw
 * /ueye_cam_nodelet/color_mode: mono8
 * /ueye_cam_nodelet/exposure: 33
 * /ueye_cam_nodelet/ext_trigger_mode: False
 * /ueye_cam_nodelet/flash_delay: 0
 * /ueye_cam_nodelet/flash_duration: 1000
 * /ueye_cam_nodelet/flip_lr: False
 * /ueye_cam_nodelet/flip_upd: False
 * /ueye_cam_nodelet/frame_rate: 40.0
 * /ueye_cam_nodelet/gain_boost: False
 * /ueye_cam_nodelet/green_gain: 1
 * /ueye_cam_nodelet/image_height: 480
 * /ueye_cam_nodelet/image_left: -1
 * /ueye_cam_nodelet/image_top: -1
 * /ueye_cam_nodelet/image_width: 640
 * /ueye_cam_nodelet/master_gain: 0
 * /ueye_cam_nodelet/pixel_clock: 25
 * /ueye_cam_nodelet/red_gain: 0
 * /ueye_cam_nodelet/subsampling: 1
 * /ueye_cam_nodelet/white_balance_blue_offset: 0
 * /ueye_cam_nodelet/white_balance_red_offset: 0

NODES
  /
    nodelet_manager (nodelet/nodelet)
    ueye_cam_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [2841]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to dc44ff70-3c41-11e5-b40e-001e06617a39
process[rosout-1]: started with pid [2854]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [2858]
process[ueye_cam_nodelet-3]: started with pid [2878]
[ INFO] [1438868929.157191461]: Initializing nodelet with 8 worker threads.
[ INFO] [1438868929.619102655]: Unable to open camera calibration file [/home/odroid/.ros/camera_info/camera.yaml]
[ INFO] [1438868936.086453623]: Synchronized UEye camera 'camera's settings and ensured compatibility with driver wrapper:
- AOI width: 752
- AOI height: 480
- AOI top-left X: 0
- AOI top-left Y: 0
[... 556 lines cut ...]
  - 
    name: pixel_clock
    value: 25
strs: 
  - 
    name: color_mode
    value: mono8
doubles: 
  - 
    name: sensor_scaling
    value: 1.0
  - 
    name: exposure
    value: 33.0
  - 
    name: frame_rate
    value: 40.0440805238
groups: 
  - 
    name: Default
    state: True
    id: 0
    parent: 0
---
```

anqixu, 2015-08-06:
Thanks for verifying the bugfix. I'll be releasing this version to rosdistro shortly.

closed 2015-08-06 by anqixu