anqixu/ueye_cam #24: Failed to set Area Of Interest (AOI)

kilmarnock, 2015-09-09:
Hi!
I am using the current version 4.60 of the ueye sdk on armhf.

I am able to use the usb_cam nodelet under the following conditions:

I have to start with image_width and image_height **half of the size of the sensor**. Setting a pixel more in any direction gives a 

"Failed to set Area Of Interest (AOI) to 752 x 480 with top-left corner at (0, 0) for [ueye]. "

When I do this via dynamic configuration, I can set the image size, but I think this is related to:

When I turn binning on (vial .launch file or dynamic), the nodelet crashes with:

**Could not start free-run live video mode for [ueye] (IS_INVALID_BUFFER_SIZE)**

Binning is supported by the camera. 

What can I do?

anqixu, 2015-09-09:
I'll need some more information before I can help.

1. fetch the latest version of ueye_cam from github, and compile from source
2. either enable debug-level message logging, or in ueye_cam/include/ueye_cam/logging_macros.hpp line 84, switch from NODELET_DEBUG_STREAM to NODELET_INFO_STREAM (and re-compile)
3. upload your ueye_cam launch file and uEye ini file (preferably on pastebin)
4. copy and paste all printouts from the ueye_cam nodelet, between the time you start the launch file, till when things crash

P.S.: I'm quite swarmed until early October, so I might not be able to provide much assistance before then

kilmarnock, 2015-09-09:
Thank you for your fast answer. Your node is the only that could support binning, so any help, even later this year, will be appreciated.

launchfile:
http://pastebin.com/hDrF9KbE
change from binning 1 to binning 2 to reproduce the error.
ini file:
http://pastebin.com/E2dpRVZ8
output:
http://pastebin.com/sf3mYbwd

Thank you

anqixu, 2015-09-09:
The problem is that the aoi queried from ids drivers already have divided the binning rate in its width and height, and then ueye_eye_driver.cpp divides those dimensions again by the binning rate. This resulted in ueye_cam allocating too small of a buffer.

I actually think the ids driver has a logical flaw, but for now, we can only fix this issue with ueye_cam...

Since I don't have a ueye cam right now, please try the following, confirm that it works, and submit a pull request:

In ueye_cam/src/ueye_cam_driver.cpp, find the following lines in the function syncCamConfig ():

// Allocate new memory section for IDS driver to use as frame buffer
  INT frameWidth = cam_aoi_.s32Width /
    (cam_sensor_scaling_rate_ * cam_subsampling_rate_ * cam_binning_rate_);
  INT frameHeight = cam_aoi_.s32Height /
    (cam_sensor_scaling_rate_ * cam_subsampling_rate_ * cam_binning_rate_);

Remove cam_binning_rate_ from the divisors and recompile.

It would be great if you could also help me test whether this issue occurs when using 2x sensor scaling, and when using 2x subsampling.

Thanks.

kilmarnock, 2015-09-09:
Almost:
It publishes images without problem. Unfortunatelly, they have the wrong size, binning=2 leads to images with width=sensor_width/4, height = sensor_height/4. Should be half, not quarter.

It leads to:
_______________
 rosrun image_view image_view image:=/ueye/image_raw
[ INFO] [1441810883.692592912]: Using transport "raw"
[ERROR] [1441810884.190198970]: Unable to convert 'mono8' image to bgr8: 'Image is wrongly formed: height * step != size  or  120 * 376 != 90240'
________________
and:
________________
rostopic echo /ueye/image_raw | grep width -A 1 -B 1
height: 120
width: 188
encoding: mono8
________________
So i cannot see the image, if it has the declared or anoter size. 

I paste you the output here:
http://pastebin.com/GDfnHK25

My cam does neither scaling nor subsampling, sorry.

Thank you so far

anqixu, 2015-09-09:
Oops I forgot about ueye_cam/src/ueye_cam_nodelet.cpp. Same issue, similar fix: find the following line, and remove the binning variable from the divisor:

ros_image_.height = cam_params_.image_height / (cam_params_.sensor_scaling * cam_params_.subsampling * cam_params_.binning);
  ros_image_.width = cam_params_.image_width / (cam_params_.sensor_scaling * cam_params_.subsampling * cam_params_.binning);

kilmarnock, 2015-09-10:
That did the trick. Thank you.
Now for that pull request: 
I have to
- create a branch
- pull the branch
- insert the changes
- upload the changes
- create a pull request

Is this what you want? No shortcut?

anqixu, 2015-09-10:
The way I know how to make pull requests is:

1. Fork repo on GitHub
2. Pull forked version
3. Commit and push changes
4. On github, there should be a button saying "make pull request"

If you like, I can take care of it. Does not matter to me.

closed 2015-09-10 by anqixu