anqixu/ueye_cam #32: IS_NOT_SUPPORTED

scott-eddy, 2016-02-24:
Hi,
I'm new to ueye cameras and am trying to use the ueye_cam package with a [UI-3590CP](https://en.ids-imaging.com/store/ui-3590cp-rev-2.html) and the 4.72 sdk of the ueyecam driver.  I'm running into an error simply running the [rgb8 launch file](https://github.com/anqixu/ueye_cam/blob/master/launch/rgb8.launch).  When running the node the package connects to the camera, resets parameters fine, and everything looks great.  However, when trying to view the image with image_view I get the following errors:
```
[ERROR] [1456345530.490563918]: Could not set free-run active-low flash output for [camera] (IS_NOT_SUPPORTED)
[ERROR] [1456345530.491632063]: Shutting down driver nodelet for [camera]
```

I'm wondering if anyone can tell me what I'm doing wrong here.  

As with #24 I've included a pastebin of my output with ROS_INFO_Streaming which can be found [here](http://pastebin.com/AH42CE1j).  I don't *think* I have a camera .ini file.

anqixu, 2016-02-24:
See if f06e0310c2029aa9d12db5a302040b658b833e12 resolves your issue.

scott-eddy, 2016-02-24:
@anqixu this fixes the issue.  Thanks for the update

closed 2016-02-24 by scott-eddy