anqixu/ueye_cam #33: Runing stereo Ueye_cam

snakehaihai, 2016-02-27:
Hi Mr Xu
I`m runing master slave demo. I deleted one camera and running 2 camera under external trigger mode.
-----------------------------------------------------------------------------------------------------------------------
Here are the facts:
I can run two camera at same time using ueyedemo
I can run 2 camera separately and sees the image using both  master slave demo(connect only 1 camera) and rgb8 demo.  By vary delay time in arduino, i can see fps change for both camera
-----------------------------------------------------------------------------------------------------------------------
Here is the problem:
But when I`m running two of them at the same time using  master slave demo, there will only be 1 working. And if i writte separate launch file by copy rgb8 demo and modify the name field. the same problem occur, only 1 camera will be running. 
It seems 2 camera are using one same pipline and somehow they are not able to share it. Have you encounter this problem before?
-----------------------------------------------------------------------------------------------------------------------
Details
    linux distro: 14.04
    ros distro:indigo
    32 bit or 64 bit os: 64
    ueye camera model:UI1221
    how many ueye cameras:2
    ids ueye driver version:uEye-Linux-4.72-64-bit
    ueye_cam git changeset: no idea 
    code execution / roslaunch procedures:  roslaunch master_slaves_rgb8.launch
    running launch file is ok. shows no error
   after view 2 image by  rosrun rviz rviz  
  output at camera driver shows the error
    execution outputs, for both successful and failed runs
dw@dw-HP-EliteBook-820-G2:~/work_space/ueye_cam/src/ueye_cam/launch$ roslaunch master_slaves_rgb8.launch
... logging to /home/dw/.ros/log/4a3a853a-dd29-11e5-afc8-94659ca2f35a/roslaunch-dw-HP-EliteBook-820-G2-24472.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://dw-HP-EliteBook-820-G2:39521/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /ueye_cam_nodelet_slave1/auto_exposure: False
 * /ueye_cam_nodelet_slave1/auto_frame_rate: False
 * /ueye_cam_nodelet_slave1/auto_gain: True
 * /ueye_cam_nodelet_slave1/auto_white_balance: True
 * /ueye_cam_nodelet_slave1/binning: 1
 * /ueye_cam_nodelet_slave1/blue_gain: 16
 * /ueye_cam_nodelet_slave1/camera_id: 1
 * /ueye_cam_nodelet_slave1/camera_intrinsics_file: slave1.yaml
 * /ueye_cam_nodelet_slave1/camera_name: slave1
 * /ueye_cam_nodelet_slave1/camera_parameters_file: 
 * /ueye_cam_nodelet_slave1/camera_topic: left_image_raw
 * /ueye_cam_nodelet_slave1/color_mode: mono8
 * /ueye_cam_nodelet_slave1/exposure: 33
 * /ueye_cam_nodelet_slave1/ext_trigger_mode: True
 * /ueye_cam_nodelet_slave1/flash_delay: 0
 * /ueye_cam_nodelet_slave1/flash_duration: 1000
 * /ueye_cam_nodelet_slave1/frame_rate: 15.0
 * /ueye_cam_nodelet_slave1/gain_boost: False
 * /ueye_cam_nodelet_slave1/green_gain: 1
 * /ueye_cam_nodelet_slave1/image_height: 480
 * /ueye_cam_nodelet_slave1/image_left: -1
 * /ueye_cam_nodelet_slave1/image_top: -1
 * /ueye_cam_nodelet_slave1/image_width: 752
 * /ueye_cam_nodelet_slave1/master_gain: 0
 * /ueye_cam_nodelet_slave1/pixel_clock: 30
 * /ueye_cam_nodelet_slave1/red_gain: 0
 * /ueye_cam_nodelet_slave1/sensor_scaling: 1.0
 * /ueye_cam_nodelet_slave1/subsampling: 1
 * /ueye_cam_nodelet_slave1/white_balance_blue_offset: 0
 * /ueye_cam_nodelet_slave1/white_balance_red_offset: 0
 * /ueye_cam_nodelet_slave2/auto_exposure: False
 * /ueye_cam_nodelet_slave2/auto_frame_rate: False
[... 105 lines cut ...]
Frame Rate (Hz):	15.003
Pixel Clock (MHz):	30
Mirror Image Upside Down:	0
Mirror Image Left Right:	0

[ INFO] [1456560529.610068289]: [slave1] set to external trigger mode
[ INFO] [1456560536.721729385]: [slave2] set to external trigger mode
[ERROR] [1456560538.721890473]: Timed out while acquiring image from [slave2] (IS_TIMED_OUT)
[ERROR] [1456560538.721934676]: If this is occurring frequently, see https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549
[ERROR] [1456560540.722178980]: Timed out while acquiring image from [slave2] (IS_TIMED_OUT)
[ERROR] [1456560540.722215219]: If this is occurring frequently, see https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549
[ERROR] [1456560542.722422993]: Timed out while acquiring image from [slave2] (IS_TIMED_OUT)
[ERROR] [1456560542.722473447]: If this is occurring frequently, see https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549
[ERROR] [1456560544.722659439]: Timed out while acquiring image from [slave2] (IS_TIMED_OUT)
[ERROR] [1456560544.722695852]: If this is occurring frequently, see https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549
[ERROR] [1456560546.722842196]: Timed out while acquiring image from [slave2] (IS_TIMED_OUT)
[ERROR] [1456560546.722887147]: If this is occurring frequently, see https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549
[ERROR] [1456560548.723238723]: Timed out while acquiring image from [slave2] (IS_TIMED_OUT)
[ERROR] [1456560548.723292800]: If this is occurring frequently, see https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549
[ERROR] [1456560550.723456031]: Timed out while acquiring image from [slave2] (IS_TIMED_OUT)
[ERROR] [1456560550.723502870]: If this is occurring frequently, see https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549
[ERROR] [1456560552.723666666]: Timed out while acquiring image from [slave2] (IS_TIMED_OUT)
[ERROR] [1456560552.723707226]: If this is occurring frequently, see https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549

    have you tried opening your cameras using ueyedemo, and seen success/failure frame captures?

progtologist, 2016-02-29:
It appears to me that your setup is dropping frames for the second camera. I experienced similar issues when I had connected two cameras in the same USB hub. When I moved one of the cameras to a USB3 interface I was able to get consistent FPS for both cameras without problems.
Careful though, IDS cameras only work on USB3 with x86 architecture processors, on ARM, with the current driver, there is no way to get good results on a USB3 controller even with one camera, at least until IDS decides to fix this.

anqixu, 2016-03-02:
Your roslaunch error messages indeed indicate that slave2 is dropping 1 frame every 2 seconds. This would however mean that you should still be successfully receiving (2*15-1) frames every 2 seconds from slave2. You should confirm this with 'rostopic hz /slave2/right_image_raw' or rosbag, rather than using rviz or image_view.

Given this infrequent rate of dropped frames, I concur with @progtologist that your cameras might be choking a single USB bus. You can try connecting the cameras in separate USB sockets on your HP EliteBook, preferably one on the USB3 bus and one on the USB2 bus. You can run the 'lsusb' terminal command to confirm that these cameras are indeed on separate buses. Also, read through https://github.com/anqixu/ueye_cam/issues/6#issuecomment-49925549 for other potential issues (e.g. long or poor USB cables).

To further debug the situation, I would recommend opening both cameras using ueyedemo, but first ensure that they are operating in external trigger mode and not freerun mode. There should be an alert icon and count in the bottom right corner of the GUI showing how many frames were dropped. You can then get more specific details on why these are being dropped, by choosing one of the menu options (I forgot its name, but it should be something like "Debug Information" or "Frame Information", etc.).

snakehaihai, 2016-03-09:
Hi anqi
I tried to run them separately with external trigger. Varying delay on arduino have effect on your rgb.launch output FPS. Running solo they are fine with each camera. So positive on external trigger.

using rostopic to echo when running 2 togerther will still result in droping frame.
I use one usb2 one usb3 and result is still the same drop frame. I ensure the camera is running at 2 different bus using lsusb.

ueyedemo external trigger is not working for some reason. as it always go in to free running mode. 
using your node running 2 camera at free running mode seems have no problem. Just that under external trigger, running 2 at the same time will result in drop frame.

Is it possible that you can try to debug using remote VNC so that you can sees the problem

anqixu, 2016-03-11:
If the USB bus throughput is not the issue, then maybe the underlying IDS driver is choking. Your guess is probably as good as mine.

If you continue having issues, I recommend contacting IDS support staff directly. Although I have some experience using and configuring uEye cameras, I am by no means an expert nor am I affiliated with IDS.

closed 2016-03-11 by anqixu

TrinhNC, 2022-03-03:
@snakehaihai I know it's been quite a long time now but I'm facing exactly the same problem here. Have you found the solution for this?

TrinhNC, 2022-03-03:
I have found the problem. There is a small bug in the **ueye_cam_nodelet.cpp**, line 213 which put the value of GPIO1 to GPIO2, so both values are always the same. I have created a pull request.

Some of my personal notice (beside others' comment) when dealing with this problem:
My wiring looks like this 

[image]

You should carefully check if the wiring is correct. I have mistakenly joined 2 red cables instead of 2 blue cables together and this timeout error also pops up.
I modified the launch file master_slaves_pwm_rgb8.launch to match the _camera ID, resolution, frame rate, etc._. So make sure these values are correct.