anqixu/ueye_cam #34: Ueye camera exposure time smaller than 1 ms

Changliu52, 2016-03-07:
Hi,
I am using the ueye_cam node with a UI-1220LE-M-GL camera. All were running perfectly in the lab.

However when I try to use it outdoor, with quite strong sun light, the node struggled to adjust the exposure to adapt this situation. 

I tried both auto and manual exposure: in manual since the exposure parameter is 'int', thus the minimum will be 1 ms, which is not small enough; in auto exposure, same thing happens, I guess it is limited to 1 ms as well.

Is this a limitation of this package? or any way I can set decimal numbers?

Thank you very much for your time,

Chang

anqixu, 2016-03-07:
- ueye_cam's exposure is internally represented as a double precision float: https://github.com/anqixu/ueye_cam/blob/master/cfg/UEyeCam.cfg#L75

- the underlying IDS API for setting exposure also takes a double:
https://github.com/anqixu/ueye_cam/blob/master/src/ueye_cam_driver.cpp#L562

- if you are referring to the sample roslaunch files, you can safely change the type of the exposure parameter to double, and pass in a decimal value.

- each camera has minimum and maximum values for exposures. If you set a value beyond this range, ueye_cam will automatically cap to those extreme values.

- when testing long exposure settings, make sure you set the clock rate to as high as possible, and reduce the frame rate to as low as possible. Once you confirm that the picture quality is good, then increase frame rate until desired, and then decrease clock rate as long as your desired frame rate can still be maintained.

closed 2016-03-07 by anqixu

Changliu52, 2016-03-07:
That's great! Thank you very much!