anqixu/ueye_cam #42: IS_INVALID_BUFFER_SIZE when enabling subsampling

scott-eddy, 2016-05-16:
Hi all,

I'm using a UI-3591LE-C-HQ with ueye_cam and am having an issue setting the sub_sampling parameter to anything other than 1.  ueyedemo allows subsampling of 2x and 4x, however ueye_cam doesn't allow for either subsampling option.  Selecting either subsampling option from rqt or the launch file outputs:

[ERROR] [1463407478.498894127]: Could not start free-run live video mode for "[camera]" (IS_INVALID_BUFFER_SIZE)
[ERROR] [1463407478.499014645]: Shutting down driver nodelet for [camera]
[Thread 0x7fffeda5d700 (LWP 5895) exited]
[Thread 0x7fffef260700 (LWP 5888) exited]
[Thread 0x7fffeea5f700 (LWP 5889) exited]
[Thread 0x7fffbaf42700 (LWP 7124) exited]
[Thread 0x7fffdbfff700 (LWP 5909) exited]
[Thread 0x7fffc0ff9700 (LWP 6025) exited]
[Thread 0x7fffc17fa700 (LWP 6022) exited]
[Thread 0x7fffc1ffb700 (LWP 6021) exited]
[Thread 0x7fffc27fc700 (LWP 6005) exited]
[Thread 0x7fffc3fff700 (LWP 5996) exited]
[Thread 0x7fffc37fe700 (LWP 5998) exited]
[Thread 0x7fffc2ffd700 (LWP 5999) exited]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[Thread 0x7fffd9a98700 (LWP 5992) exited]
[Thread 0x7fffda299700 (LWP 5994) exited]
[Thread 0x7fffd9297700 (LWP 5993) exited]
[Thread 0x7fffe57fa700 (LWP 5904) exited]
[Thread 0x7fffe4ff9700 (LWP 5905) exited]
[Thread 0x7fffed25c700 (LWP 5897) exited]
[Thread 0x7fffe5ffb700 (LWP 5903) exited]
[Thread 0x7fffe6ffd700 (LWP 5901) exited]
[Thread 0x7fffe67fc700 (LWP 5902) exited]
[Thread 0x7fffeca5b700 (LWP 5898) exited]
[Thread 0x7fffe77fe700 (LWP 5900) exited]
[Thread 0x7fffe7fff700 (LWP 5899) exited]
[Thread 0x7fffba741700 (LWP 6391) exited]
[Thread 0x7fffee25e700 (LWP 5890) exited]
[Inferior 1 (process 5870) exited normally]

It looks like a similar (IS_INVALID_BUFFER_SIZE) problem was fixed for binning in https://github.com/anqixu/ueye_cam/commit/2bee82e54e0a303be54e0c949465b968f59a476e.  Any advice on how to fix the issue for subsampling?

anqixu, 2016-05-16:
Unfortunately my current uEye cam does not support subsampling, so we'll have to look into this together.

**Diagnosis steps:**

1. fetch the latest version of ueye_cam from github
2. either enable debug-level message logging, or in ueye_cam/include/ueye_cam/logging_macros.hpp line 84, switch from NODELET_DEBUG_STREAM to NODELET_INFO_STREAM (and re-compile)
3. compile source and roslaunch (ensure beforehand that your catkin workspace is overlaid on top of /opt/ros/...)
4. grab images from the camera successfully, then change subsampling via rqt_reconfigure
5. copy all output and post somewhere (pastebin)

**Potential fix:**

Just as in https://github.com/anqixu/ueye_cam/commit/2bee82e54e0a303be54e0c949465b968f59a476e, try removing "cam_subsampling_rate_"  and "cam_params_.subsampling" from the highlighted lines. If it works, I'd welcome a pull request or a reply of acknowledgement.

P.S.: please make sure that you are using the latest IDS drivers.

scott-eddy, 2016-05-16:
removing those two lines fixes the problem, in the driver cam_aoi_.s32Width/Height change automatically with subsampling factor.

scott-eddy, 2016-05-16:
I added PR #43 for a fix.

PR #43 (2016-05-16): Fix buffersize for subsampled images
fixes #42
closed 2016-05-16 by anqixu