anqixu/ueye_cam #48: stereo camera

Annaiyan, 2016-06-30:
I'm trying to use the ueye_cam package to launch a two independent ueye camera as stereo camera, Because i want to do stereo processing which needs identical time stamps for both the camera's.
I have modified the master_slave launch file and executed with two camera's.
One camera is failed to intialize and the error is like below:

> ERROR] [1467313081.966489221]: Could not open UEye camera ID 2 (IS_CANT_OPEN_DEVICE)
> [ERROR] [1467313081.966554728]: Failed to initialize UEye camera 'slave'
> 
I would like to launch the both camera's for stereo_image_processing.
Any suggestion or ideas for this issue?
Thanks in advance!

anqixu, 2016-06-30:
As the error message suggests, the specified camera ID is wrong.
To see/set what are the camera IDs for your devices, run 'ueyecameraconfig'.

Also, to make master_slave.launch work, you will need to connect the flash output pin from the master camera to the trigger input pin to the slave camera. See [comments here](https://github.com/anqixu/duino_projects/blob/master/ueye_cam_ext_trigger/ueye_cam_ext_trigger.ino#L84) for more details.

In addition to the above, you'll also either need to logically negate the trigger signal (e.g. using an arduino device + code linked above), or change [the code in ueye_cam](https://github.com/anqixu/ueye_cam/blob/master/src/ueye_cam_driver.cpp#L843) so that the slave triggers on IS_SET_TRIGGER_LO_HI instead of IS_SET_TRIGGER_HI_LO.

Annaiyan, 2017-10-10:
Hello,
Recently i have connected the two ueye camera (UI-1221LE-C) like mentioned above i.e from pin4 to pin 3 using a simple wire. Also i had changed the code from S_SET_TRIGGER_HI_LO to IS_SET_TRIGGER_LO_HI. but when i launch the _master_slaves_rgb8.launch_ file it is crashing always! Below is the error message

```
[ERROR] [1507648554.977615481]: Could not enable falling-edge external trigger mode for [right] (IS_NO_SUCCESS)
[ERROR] [1507648554.977657027]: Shutting down driver nodelet for [right]
```

what could be solution to avoid the crashing?

anqixu, 2017-10-11:
From your pasted error message:
  [ERROR] [1507648554.977615481]: Could not enable falling-edge external trigger mode for [right] (IS_NO_SUCCESS)

The text part comes from the code I wrote from ueye_cam, while the part in the parentheses `IS_NO_SUCCESS` comes from IDS drivers. Based on this message, I believe that your particular camera **does not support external trigger settings**.

To confirm this, use ueyedemo to connect to your camera, then go in its options, and try to toggle the external trigger settings, and see if that works.

Either way, since I don't have your camera, I cannot help you debug further on this matter. Please contact IDS for more information.

closed 2017-10-11 by anqixu