anqixu/ueye_cam #56: Could not load [camera]'s sensor parameters file

cowboygary, 2017-06-01:
Hi

When I launch rgb8.launch, i get this error:

[ WARN] [1496331955.280376918]: Could not load [camera]'s sensor parameters file ~/catkin_ws/src/ueye_cam/camera_conf/ (IS_FILE_WRITE_OPEN_ERROR)
[ WARN] [1496331955.320810776]: [camera] does not support internal image scaling

I get the ini file from ueyedemo. 

I need to get the camera calibrated.

My question is:
1. why i got this error?
2. why in INI file, there's no intrinsic parameter? How can I set the intrinsric parameter?

Please help me.
I'm new to ROS programming.

anqixu, 2017-06-02:
You need to pass the absolute path to the ini file in the following ROS param within the launch file, e.g.:

<param name="camera_parameters_file" type="string" value="/PATH/TO/MY/FILE.INI" />

Also, the ini file only contains camera configurations and does not contain any intrinsics. You should check out the following link for more details on camera calibration in ROS. Note that ueye_cam does support the camera_calibration package:

http://wiki.ros.org/camera_calibration/Tutorials

closed 2017-06-02 by anqixu

cowboygary, 2017-06-07:
Hi
thanks for your quick answer.

I enter this command: rostopic echo /camera/camera_info

then, I get this :

_---
header: 
  seq: 811
  stamp: 
    secs: 1496841589
    nsecs: 627545999
  frame_id: /camera
height: 480
width: 640
distortion_model: ''
D: []
K: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False_
__

How I can set all these parameters?
Thank you

anqixu, 2017-06-07:
You need to calibrate the camera first using camera_calibration node first (Calibrate -> Save), and then /camera/camera_info will be populated properly.