anqixu/ueye_cam #57: Add xenial arm64 to ROS Jenkins buildfarm

plusk01, 2017-07-07:
Can we add a 64-bit ARM (`arm64`) `ueye_cam` binary to the build farm?

This would make it easier to run on embedded devices such as the Jetson TX1 and TX2.

anqixu, 2017-07-10:
I support this, but am too swamped to make this change. Please make a pull request and test it with a camera, and I'll approve after testing compilation on my TX2 (i also don't have any ueye cams on hand now)

anqixu, 2017-07-10:
note that you will need to download the aarch64 IDS drivers, package them to a zip, upload them to ros servers (maybe email Dirk Thomas for help?), and update CMakeLists.txt to use the appropriate ARCH flag and corresponding zip file. See an existing zip file linked in the CMakeLists for which files should be packaged.

standmit, 2018-06-21:
Does this package works on Jetson TX2 currently?

anqixu, 2018-06-21:
Still didn't have time to test this nor do I have a uEye camera, sorry :(

standmit, 2018-06-22:
It works. I launched it 'as is' on Jetson TX2 and got 1440x810 RGB8 at 90 FPS.

JadTawil-theonly, 2020-11-20:
@plusk01 is support for arm64 available yet?

anqixu, 2021-01-10:
PR submitted for rosdistro

closed 2021-01-10 by anqixu