anqixu/ueye_cam #58: no recovery after timeout

kilmarnock, 2017-07-14:
Hi!
I have a problem you probably could give me an advice.
I observe timeouts with my UI1221LE-M-GL. I read #6.  I tried ARM, amd64, usb2 + usb3. I changed the pixel_clock settings, all in auto mode. I bought an pcie2usb adapter. I measure the Bus > 5 Volt.
The cam gets more stable when I disconnect each and every usb device.

Each time the timeout occurs, the node is lost, **it does not recover from timing out**, it basically loops in the timeout (`UEyeCamDriver::processNextFrame: is_err == IS_TIMED_OUT`). Until now, I had to disconnect the cam and restart the node. I thought that the hardware crashes for some reason. 

Today I found this trick:
`echo "0" > /sys/bus/usb/devices/usb2/authorized `
-> /dev/ueye disappears
`echo "1" > /sys/bus/usb/devices/usb2/authorized`
-> /dev/ueye reappears
Then I can continue until the next timeout.

Correct me if I am wrong, but it does not seem to be hardware issue.

ueyecamerademo recovers and just counts missed frames.

Do you have a clue for me where to look now?

kilmarnock, 2017-07-18:
after the timeout, the camera hangs even after a restart of the ros driver and the ueye daemon. 
`EyeCamDriver::setFreeRunMode()` leads to `!freeRunModeActive()` and `setStandbyMode()` where it stays forever.

anqixu, 2017-10-19:
Sorry I'm still not able to replicate your results, and given that I don't have access to uEye cameras, so it will be hard to debug.