anqixu/ueye_cam #60: [UI-3131LE-M-GL] Could not set flash parameter info

artifactz, 2017-10-26:
I have the UI-3131LE monochrome camera and it fails claiming

    [ INFO] [1509019730.437946409]: Initializing nodelet with 4 worker threads.
    [ INFO] [1509019730.474593811]: Unable to open camera calibration file [/home/dai/.ros/camera_info/camera.yaml]
    [ WARN] [1509019731.177803826]: Could not load [camera]'s sensor parameters file /home/dai/.ros/camera_conf/camera.ini (IS_FILE_WRITE_OPEN_ERROR)
    [ WARN] [1509019731.318777563]: [camera] does not support internal image scaling
    [ INFO] [1509019731.566051275]: UEye camera [camera] initialized on topic /camera/image_raw
    Width:			640
    Height:			480
    Left Pos.:		-1
    Top Pos.:		-1
    Color Mode:		rgb8
    Subsampling:		1
    Binning:		1
    Sensor Scaling:		1
    Auto Gain:		1
    Master Gain:		0
    Red Gain:		0
    Green Gain:		1
    Blue Gain:		16
    Gain Boost:		0
    Auto Exposure:		1
    Exposure (ms):		33
    Auto White Balance:	1
    WB Red Offset:		0
    WB Blue Offset:		0
    Flash Delay (us):	0
    Flash Duration (us):	1000
    Ext Trigger Mode:	0
    Auto Frame Rate:	0
    Frame Rate (Hz):	29.9902
    Output Rate (Hz):	0
    Pixel Clock (MHz):	30
    Mirror Image Upside Down:	0
    Mirror Image Left Right:	0
    
    [ERROR] [1509019731.566989662]: Could not set flash parameter info for [camera] (IS_NOT_SUPPORTED)
    [ERROR] [1509019731.601301121]: Could not set flash parameter info for [camera] (IS_NOT_SUPPORTED)
    [ERROR] [1509019731.601351488]: Shutting down driver nodelet for [camera]

I've tried without any effect:
* changing color mode to `mono8`
* changing flash duration to `0` or `-1`
* changing image size to 800 x 600
* disabling auto exposure

I've also tried enabling `ext_trigger_mode`:

    Timed out while acquiring image

I assume my camera does not support setting flash parameters, so not setting them at all could solve the issue. How can I do that?
Any help is appreciated.

anqixu, 2017-10-28:
See if this commit fixes your issue.

2ed208eef90e5a082557fe878cff90ede58880f9

artifactz, 2017-11-02:
Yes! Very nice, thank you. :+1: 
However, the error message still appears (twice), so maybe change it to a warning..?

anqixu, 2017-11-13:
done

closed 2017-11-13 by anqixu