anqixu/ueye_cam #62: Camera imu synchronisation

Changliu52, 2017-11-04:
Hi @anqixu,

Thank you for the neat project.
I am trying to develop hardware sync between imu and camera.

I am thinking to use an arduino with mpu6050 and generate external trigger signal for a ueye cam. Those are all cool, but I need a way to change the timestamp for the output image frame, in order to have the same time frame with the triggering imu message.

Is there a way, that is implemented to do that?

Thank you for your time.
Chang

anqixu, 2017-11-10:
If you want the IMU + camera to be perfectly synchronized in real-life, then that's only possible up to a certain temporal accuracy based on the length of wires you chooose, and the timing accuracy of the microcontroller you choose. Generally speaking, it's really hard to align two slave devices to be triggered within a few us of each other.

However, if you just want to process synchronized IMU and camera messages together, have a look at [TimeSynchronizer](http://wiki.ros.org/message_filters#Time_Synchronizer).

And if you really really want the IMU and camera message timestamps to be identical, you can have an intermediary node use TimeSynchronizer, then overwrite camera.header.stamp with imu.header.stamp.

closed 2017-11-10 by anqixu

Changliu52, 2017-11-10:
Great! Thanks.

Chang