anqixu/ueye_cam #64: Master Slave mode for synchronization

ank95, 2018-03-27:
Hello Mr. Xu

First of all i'd like to thank you for creating such a useful package for ROS intergration with Ueye. It has been really helpful with my project. I am currently trying to create a setup that uses 2 Ueye cameras for synchronisation. I want to use the master slave configuration with Cam1 as the master in free-run mode and Cam2 as the slave. 

Due to lack of hardware exprience, I was hoping if you could give me a few pointers on the hardware setup required. The questions i have are as follows:
1. On the hardware side i have to connect Flash output of Master directly to the trigger input of Slave, run the Master in free run mode and Slave in Ext_trigger mode?
2. You mentioned that i need to ensure on the software side that the slave camera triggers on rising edge (LO_HI) rather than on falling edge (HI_LO). Where and exactly what changes should be done in the code? 

Thanks in advance.

anqixu, 2018-03-27:
Hi ank95. Glad to hear you finding this package useful.

To answer your questions:
1. that sounds correct
2. I believe so

Unfortunately I haven't had access to these cameras for a while so I can't test it out directly. Let me know how things go.

ank95, 2018-04-03:
Thank you for the response Mr.Xu. I'll post an update when i have the hardware setup. I'm also trying to get in touch with IDS for further support.

ank95, 2018-05-25:
Hello Mr. Xu

Apologies for the late update. I just received all the necessary hardware yesterday. I followed you suggestions together with [this](https://en.ids-imaging.com/tl_files/downloads/support_notes/AN_AB.1.0030_Synchronizing-Image-Acquisition-uEye-Cameras_EN.pdf) document from IDS to make the hardware connections. 

Since my camera supports both Rising & Falling edge triggers i was to use the package directly without needing to invert the IS_SET_TRIGGER parameter. Thank you for the help.

closed 2018-05-25 by ank95