anqixu/ueye_cam #70: camera Id

JadBatmobile, 2019-01-04:
I am using two camera nodelets in a single manager (i modified rgb8.launch).

How can i ensure that the same cameras will have the same IDs each time i launch the driver.

Sometimes camera_1/image.. refers to the camera on the right side of the robot, and sometimes left. Same for camera_2/image...

JadBatmobile, 2019-01-04:
is there a way i can specify the hardware id of the camera to the ros driver?

JadBatmobile, 2019-01-15:
Can you please elaborate on how to use eyesetid or ueyecameramanager. Are these interfaces provided by your package?

anqixu, 2019-01-15:
These are software binaries provided by IDS Imaging when you installed their uEye SDK.

JadBatmobile, 2019-01-15:
yes thank you, i found and used.