anqixu/ueye_cam #71: REQUIRED process [check_ueye_api-1] has died!

liu-zg15, 2019-01-08:
Hi, this is my first time to use ueye camera. My camera is **UI324xLe-M**. I installed the **driver 4.91.0001**. All things are good when I tried to launch the camera by camera Manager. So I tried to launch the camera by ROS. After catkin_make the package. I entry the **sudo /etc/init.d/ueyeusbdrc start** to start the driver first. Then launch the roscore in a new terminal. At last I cd to catkin_ws/src/ueye_cam/launch/ and run **roslaunch rgb8.launch** in terminal. But error happens:
```
process[ueye_cam_nodelet-3]: started with pid [4582]
[ INFO] [1546972685.473706366]: Initializing nodelet with 4 worker threads.
[ INFO] [1546972685.498840234]: Unable to open camera calibration file [/home/gone/.ros/camera_info/camera.yaml]
================================================================================REQUIRED process [check_ueye_api-1] has died!
process has died [pid 4579, exit code -6, cmd /home/gone/catkin_ws/devel/lib/ueye_cam/check_ueye_api __name:=check_ueye_api __log:=/home/gone/.ros/log/3b7297de-1374-11e9-873f-5800e34de6a7/check_ueye_api-1.log].
log file: /home/gone/.ros/log/3b7297de-1374-11e9-873f-5800e34de6a7/check_ueye_api-1*.log
Initiating shutdown!
================================================================================
[ueye_cam_nodelet-3] killing on exit
[nodelet_manager-2] killing on exit
[check_ueye_api-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

I don't know why this happens, can anybody help me?

anqixu, 2019-01-14:
I don't know why you are getting this error. Exit code -6 *might* be due to Linux System Error code 6, which apparently says ["No such device or address"](http://www-numi.fnal.gov/offline_software/srt_public_context/WebDocs/Errors/unix_system_errors.html). If that's truly the problem, then you should make sure that your camera is properly connected, powered on, and the USB cable is stable, USB hub available, etc.

Otherwise, since I can't replicate your error on my setup, you will have to debug which line of [this code](https://github.com/anqixu/ueye_cam/blob/master/src/check_ueye_api.cpp) (e.g. via `gdb`) is crashing, in order to dig deeper.

Good luck.

closed 2019-01-14 by anqixu

ohmsol, 2019-07-17:
I get the same error message. Camera is a UI-3013XC-C, installed the software suite and started the driver (sudo /etc/init.d/ueyeusbdrc start).

Did you manage to find a solution?