anqixu/ueye_cam #73: adjust exposure parameter

dean36995, 2019-01-29:
Sorry, I just adjust the exposure parameter, but it doesn't work. I used ROS kinetic version to change the parameter in "bayer_rggb_image_proc.launch" file, but only the image size and color mode can change. Do you have any method to control the image exposure? Thank you very much.

anqixu, 2019-01-29:
Maybe your uEye camera does not support changing exposure? Also, make sure to turn off auto-exposure + auto-framerate, and note also that the upper limit of exposure value is bound by the pixel clock and frame rate of your camera.

To verify the above, try changing settings using IDS's `ueyedemo` executable. If it works there, please paste all messages (INFO, WARNING, ERROR) that you get when you `roslaunch ...`.

dean36995, 2019-01-30:
1. 
I can “use ueye properties” tool when I type the command “ueyedemo” in terminal, then I cancel the “auto” selection which below the “Exposure”, and I can adjust the Exposure parameter by move the “Exposure” bar.
As Feature.1 and Feature.2. To prove the uEye camera can support changing exposure.
 
[image]

Feature. 1
 
[image]

Feature. 2
2.
I change the parameter “width:1280”, “height:720”, “auto_exposure: False”, “auto frame_rate: False” in “xs_bayer_rggb.lunch” file, and run the command “roslaunch ueye_cam xs_bayer_rggb.launch”. As below.

roslaunch ueye_cam xs_bayer_rggb.launch 
... logging to /home/ubuntu/.ros/log/f6755f08-243b-11e9-bce0-00044ba7ed11/roslaunch-tegra-ubuntu-30662.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:45833/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /ueye_cam_nodelet/auto_exposure: False
 * /ueye_cam_nodelet/auto_frame_rate: False
 * /ueye_cam_nodelet/auto_gain: True
 * /ueye_cam_nodelet/auto_white_balance: True
 * /ueye_cam_nodelet/binning: 1
 * /ueye_cam_nodelet/blue_gain: 16
 * /ueye_cam_nodelet/camera_id: 0
 * /ueye_cam_nodelet/camera_intrinsics_file: 
 * /ueye_cam_nodelet/camera_name: camera
 * /ueye_cam_nodelet/camera_parameters_file: /home/ubuntu/catk...
 * /ueye_cam_nodelet/camera_topic: image_raw
 * /ueye_cam_nodelet/color_mode: bayer_rggb8
 * /ueye_cam_nodelet/exposure: 33
 * /ueye_cam_nodelet/ext_trigger_mode: False
 * /ueye_cam_nodelet/flash_delay: 0
 * /ueye_cam_nodelet/flash_duration: 1000
 * /ueye_cam_nodelet/flip_lr: False
 * /ueye_cam_nodelet/flip_upd: False
 * /ueye_cam_nodelet/frame_rate: 30.0
 * /ueye_cam_nodelet/gain_boost: False
 * /ueye_cam_nodelet/green_gain: 1
 * /ueye_cam_nodelet/image_height: 720
 * /ueye_cam_nodelet/image_left: 0
 * /ueye_cam_nodelet/image_top: 0
 * /ueye_cam_nodelet/image_width: 1280
 * /ueye_cam_nodelet/master_gain: 0
 * /ueye_cam_nodelet/output_rate: 0.0
 * /ueye_cam_nodelet/pixel_clock: 30
 * /ueye_cam_nodelet/red_gain: 0
 * /ueye_cam_nodelet/sensor_scaling: 1.0
 * /ueye_cam_nodelet/subsampling: 1
 * /ueye_cam_nodelet/white_balance_blue_offset: 0
 * /ueye_cam_nodelet/white_balance_red_offset: 0

NODES
  /
    nodelet_manager (nodelet/nodelet)
    ueye_cam_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [30672]
ROS_MASTER_URI=http://localhost:11311
[... 86 lines cut ...]

5. 
I also change the parameter in launch file.
Image “width:1280”, “hight:720”, “auto exposure: False”, “auto frame rate: False”.
Feature. 4 parameters is “exposure:1” “frame rate: 30”.
Feature. 5 parameters is “exposure:33” “frame rate: 30”.
These two features like the same.

 
[image]

Feature. 4
 
[image]

Feature. 5
6. Other information:
Camera_ini
 
[image]

I try to run the other launch file. “bayer_rggb_image_proc”. But the performance like the “xs_bayer_rggb.lunch” file.

If there is not enough information, please let me know. 
Thank you very much.

dean36995, 2019-03-07:
Sorry, excuse me.
this kind of camera can set "fixed" focal length?

anqixu, 2019-03-07:
I don't think so, but you should check with the camera vendor to be sure.

nullket, 2020-12-20:
As this issue has stalled, I am closing it to clean up the repo a bit. Hope your issue was solved, if not feel free to open a new issue and include debugging information.

closed 2020-12-20 by nullket