anqixu/ueye_cam #76: IDS Ueye 4.92 munmap_chunk invalid pointer in ros::init

devrite, 2019-09-09:
Hi everyone!

We used a  `ros:melodic-perception-bionic` the docker image (with the necessary dependencies installed):
```
#Digest: sha256:2a8412596201e7be8a181f4523d00446f6e0a37e659eee4f6a2898139d11972a

docker pull ros:melodic-perception-bionic
docker run --rm --name ids-4.92-ros-error -v$HOME/Downloads/ueye_4.92.0.0_amd64.run:/ueye_4.92.0.0_amd64.run:ro -it  ros:melodic-perception-bionic

apt-get update && apt-get upgrade
apt-get -y install wget software-properties-common build-essential wget curl cmake git rsync gdb

/ueye_4.92.0.0_amd64.run
mkdir /build
cd /build
git clone https://github.com/anqixu/ueye_cam.git
cd ueye_cam
mkdir debug
cd debug
cmake -DCMAKE_BUILD_TYPE=Debug ..
make -j4
source devel/setup.bash
rosrun ueye_cam check
```

Using gdb and catch throw you may print a backtrace (error in ros::init -> file log).

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007fc255fda801 in __GI_abort () at abort.c:79
#2  0x00007fc256023897 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fc256150b9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007fc25602a90a in malloc_printerr (str=str@entry=0x7fc2561527a8 "munmap_chunk(): invalid pointer") at malloc.c:5350
#4  0x00007fc256031ecc in munmap_chunk (p=0x55a6778ba5a0) at malloc.c:2846
#5  __GI___libc_free (mem=0x55a6778ba5b0) at malloc.c:3117
#6  0x00007fc2579595d0 in ros::file_log::init(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&) ()
   from /opt/ros/melodic/lib/libroscpp.so
#7  0x00007fc257981821 in ros::init(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int) () from /opt/ros/melodic/lib/libroscpp.so
#8  0x00007fc25798416c in ros::init(int&, char**, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int) () from /opt/ros/melodic/lib/libroscpp.so
#9  0x000055a677146345 in main (argc=1, argv=0x7ffc53e989a8) at /build/ueye_cam/src/check_ueye_api.cpp:45
```

Best regards, 

Markus

devrite, 2019-09-09:
Using lldb -r . -s libueye it looks like ros::init jumps into the ueye implementations of boost stuff (especially filesystem stuff). So there might be a version conflict in the used boost versions.

loooph, 2019-10-03:
Might be late but i had the same issue and have resolved it in #77 .

devrite, 2019-10-08:
As far as I can see it is not being linked at link time. Check_ueye_api links it at runtime, which is fine if you want to check if it can be loaded.

The problem is that the ueye node or our own camera node fails as soon as ros::init is being called. Check ueye is only the smallest reproducible example of the error.

As far as I can see ueye API exports it's own boost functions, which are incompatible with the system boost. In my opinion the ueye api needs to be rebuilt. Older versions of ueye still work.

I think without help from IDS this can not be solved... 😑

devrite, 2019-10-08:
Sorry yes it was being linked at link time, but the linker kicked it since no function was being called.

At least in my example the library was not loaded when entering main only after dlopen. If it was being linked at link time it should be present when entering main usually.

devrite, 2019-10-09:
Sorry, I checked today. It was being linked. The issue still remains in other context where the library is not being loaded at runtime dynamically (in the nodelet context it will be, hence it works, right now ...). 

The libary was probably not kicked because a function is being used (like the above boost export).I think the linker usually checks this.

For check_ueye_api the issue is solved by not linking it as you suggest. But for any node that uses the library via build time linking will fail as check_ueye_api

ChrisSchulte, 2019-10-29:
Hi everyone,

I'm having the same troubles with ROS Kinetic on Ubuntu 16.04 LTS with IDS SDK version 4.91.1. Same behaviour as above, a munmap_chunk() : invalid pointer crash as soon as a build time link with the ueye_api exists, even if no call to any ueye function is made in the executable.
Every executable using ueye without libroscpp.so works flawlessly.

Chris

Simen-Thys, 2019-12-05:
Perhaps we could use [Implib.so](https://github.com/yugr/Implib.so) to hide the unnecessary boost symbols. I haven't tested this myself though.

devrite, 2024-07-16:
Not verified if it is solved upstream by now. Closing due to lack of activity.

closed 2024-07-16 by devrite