anqixu/ueye_cam #78: exit code -6

alduxvm, 2019-10-24:
Hi!

I'm trying to use this package with ubuntu 18.04 and ros melodic. I installed the IDS software suite (uEye-Linux-4.92.0-64) and I try the camera first with the ueye_demo app, the camera works.

Then I try to launch the rgb8 and I'm getting this error:

```roslaunch src/ueye_cam/launch/rgb8.launch 
... logging to /home/aldux/.ros/log/637f0ea2-f671-11e9-8539-5c5f67f52409/roslaunch-x1-26930.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://x1:43815/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /ueye_cam_nodelet/auto_exposure: True
 * /ueye_cam_nodelet/auto_frame_rate: False
 * /ueye_cam_nodelet/auto_gain: True
 * /ueye_cam_nodelet/auto_white_balance: True
 * /ueye_cam_nodelet/binning: 1
 * /ueye_cam_nodelet/blue_gain: 16
 * /ueye_cam_nodelet/camera_id: 0
 * /ueye_cam_nodelet/camera_intrinsics_file: 
 * /ueye_cam_nodelet/camera_name: camera
 * /ueye_cam_nodelet/camera_parameters_file: 
 * /ueye_cam_nodelet/camera_topic: image_raw
 * /ueye_cam_nodelet/color_mode: rgb8
 * /ueye_cam_nodelet/exposure: 33
 * /ueye_cam_nodelet/ext_trigger_mode: False
 * /ueye_cam_nodelet/flash_delay: 0
 * /ueye_cam_nodelet/flash_duration: 1000
 * /ueye_cam_nodelet/flip_lr: False
 * /ueye_cam_nodelet/flip_upd: False
 * /ueye_cam_nodelet/frame_rate: 30.0
 * /ueye_cam_nodelet/gain_boost: False
 * /ueye_cam_nodelet/green_gain: 1
 * /ueye_cam_nodelet/image_height: 480
 * /ueye_cam_nodelet/image_left: -1
 * /ueye_cam_nodelet/image_top: -1
 * /ueye_cam_nodelet/image_width: 640
 * /ueye_cam_nodelet/master_gain: 0
 * /ueye_cam_nodelet/output_rate: 0.0
 * /ueye_cam_nodelet/pixel_clock: 25
 * /ueye_cam_nodelet/red_gain: 0
 * /ueye_cam_nodelet/sensor_scaling: 1.0
 * /ueye_cam_nodelet/subsampling: 1
 * /ueye_cam_nodelet/white_balance_blue_offset: 0
 * /ueye_cam_nodelet/white_balance_red_offset: 0

NODES
  /
    check_ueye_api (ueye_cam/check_ueye_api)
    nodelet_manager (nodelet/nodelet)
    ueye_cam_nodelet (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [26940]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 637f0ea2-f671-11e9-8539-5c5f67f52409
process[rosout-1]: started with pid [26951]
started core service [/rosout]
process[check_ueye_api-2]: started with pid [26957]
process[nodelet_manager-3]: started with pid [26959]
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_M_create
process[ueye_cam_nodelet-4]: started with pid [26961]
[ INFO] [1571930247.974648798]: Initializing nodelet with 8 worker threads.
[ INFO] [1571930247.999676539]: Unable to open camera calibration file [/home/aldux/.ros/camera_info/camera.yaml]
================================================================================REQUIRED process [check_ueye_api-2] has died!
process has died [pid 26957, exit code -6, cmd /home/aldux/ros-ueye-test/devel/lib/ueye_cam/check_ueye_api __name:=check_ueye_api __log:=/home/aldux/.ros/log/637f0ea2-f671-11e9-8539-5c5f67f52409/check_ueye_api-2.log].
log file: /home/aldux/.ros/log/637f0ea2-f671-11e9-8539-5c5f67f52409/check_ueye_api-2*.log
Initiating shutdown!
================================================================================
[ueye_cam_nodelet-4] killing on exit
[nodelet_manager-3] killing on exit
[check_ueye_api-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Any idea about this issue?

Thanks a lot!

devrite, 2019-10-30:
> `process[check_ueye_api-2]: started with pid [26957] -> rocess has died [pid 26957]`

Since check ueye node fails it is probably the same error as in #76 and #77. #77 fixes this for this package.

Beware though if you plan to use IDS Ueye directly with ros (no plugin load in your own code) you will experience the same error, since it looks like ueye exposes some boost api functions via their library, which conflict with the one ros wants to use.

nullket, 2020-12-20:
I am cleaning this repo a bit and this is a stalled issue I hope you do not mind if I close it. As mentioned the error should be fixed in the linked PR, thanks!

closed 2020-12-20 by nullket