anqixu/ueye_cam #84: Overexposure

RGring, 2020-07-31:
Hi. Thanks for your work!

I was wondering if you (or anyone else) experienced a strong overexposure (outdoors as well as indoors), when setting auto exposure. This also leads to very unnatural colors. It almost seems, that the auto exposure algorithm is not configured right?

Here is an example: Upper image (ueye), lower image (from another camera and more realistical).

[image]

anqixu, 2020-07-31:
I've seen that issue myself, but I'm no camera/lens expert, so I'm not certain whether this is a ueye-specific issue or that perhaps it can be tuned away. In any case, ueye_cam simply passes through settings from the uEye IDS driver. Are you able to reproduce similar results while using IDS's ueyedemo software? If so I recommend you contact IDS for further support.

RGring, 2020-08-03:
Thanks for sharing your experience! In case others have the problem. I played around with the demo software and found out that setting the software gamma to 128 helped quite a lot.

closed 2020-08-03 by RGring