anqixu/ueye_cam #91: exit camera with code 3

sepid91, 2020-10-13:
Hi,
I use  camera ui-5240se-c-hq-rev2  and kubuntu 20.04 with idscameramanager 4.93 but it gives me code 3 (IS_CANT_OPEN_DEVICE). how can i fix it?
I tested the camera with windows and it worked. I tested all version of suit and i got the same error.

nullket, 2020-10-20:
This is most likely not a problem of this driver.

Speaking for me: 4.93 & Kubuntu 20.04 & latest git clone from here works fine with a UI-5270CP-CHQ-R2. My camera seems to be fairly comparable to yours (in terms of what is relevant for this driver). 

`IS_CANT_OPEN_DEVICE` usually means, that the camera is busy otherwise. This mostly happens when you have the camera opened within `uEye demo` (not the ids cameramanager, the ids program that just shows the camera image) while trying to launch the ros file. The camera can only be accessed by one program at the same time (means you need to close `uEye demo`).

If that is not the problem check the following:
- Does the Camera give an image within  `uEye demo` in Linux?
- Are you sure, that your network connection to the camera is correct configured (Normally this means a static IP in Linux)

nullket, 2020-12-12:
Closing due to no response

closed 2020-12-12 by nullket