anqixu/ueye_cam #93: Problem with UI1007XS

sararht, 2020-11-10:
Hi,

The package works fine for me with older ueye XS cameras. However, with the UI1007XS-C model, when executing any launch, it gives me an error.

For example, when launching xs_bayer_rggb.launch, the resulting error is as follows:

```
[ERROR] [1604999300.440341174]: Could not set color mode of [camera] to mono8 (IS_INVALID_COLOR_FORMAT: 6 / 'mono8'). switching to default mode: mono8
[ERROR] [1604999300.440452813]: Could not set color mode of [camera] to mono8 (IS_INVALID_COLOR_FORMAT: 6 / mono8)
[ERROR] [1604999300.440595742]: Failed to initialize [camera]
```

However, I can open both cameras normaly with idscameramanager.

Any help? Thanks in advance.

nullket, 2020-12-21:
According to the ids manual the UI1007XS has the specialty of being able to calculate the color information directly on the camera and transfer it via YCbCr-Format. Most other cameras transfer the unbayered raw image (more or less) and then the IDS Ueye API transfers this image into another color mode. Nonetheless, the manual says that the color mode can be converted by the ueye API for the UI1007XS. I assume this might not be possible for an unbayered format like xs_bayer_rggb.launch would create. How does the normal rgb8.launch work? 

Nonetheless, I am afraid that this camera might not work with the current driver. It might need the ueye API function `is_Convert()` instead of `is_SetColorMode()` as the color is already calculated by the camera itself. But I am not sure as the IDS manual is not quite specific here. 

Unfortunately, supporting YCbCr-Format is not really possible without a good portion of development (introducing `is_convert` and figuring out when to use it) as ROS does not seem to support the YCbCr-Format. So a simple "adding YCbCr-Format as a mode to to be selected in the driver" would not help as then ros would not work with this colorformat image.

As mentioned, this is just the result of my quick research. So not everything might be correct.

sebastianfast, 2021-05-26:
@sararht @nullket Have you made any progress on this? I have stumbled over the same problem, only that back on version `1.0.16` I was able to use the camera. Now I'm forced to use Ubuntu 20.04, which forces me to use the uEye-driver 4.94, which forces me to use the version `1.0.19` of this packages. Here I am now, stuck.

nullket, 2021-05-26:
I do not know why 1.0.16 worked, this was before my time here. 

I do not have access (and the time for the development) to such a camera. But I am happy to help you here on github with feedback and point you in the right directions to develop you own branch supporting those cameras (of course this should be merged here as well).

sebastianfast, 2021-05-26:
Thanks for your quick response and your offer to guide me through the development! Unfortunately we don't have the time for this, too. However, should we decide to work on this, I will make sure the code will find its way here!

Sr4l, 2021-11-15:
If you set the color_mode to "bgr8" in the "rgb8.launch" file everything seems to work with:
- camera UI1007XS-C
- ids drivers 4.95
- Ubuntu 20.04
- ROS Noetic

nullket, 2021-11-15:
Yes, you are right. Thanks for the addition. According to the manual the camera does support the following profiles.The following are IDS names, their corresponding launch file name can be found here https://github.com/anqixu/ueye_cam./blob/749ee7ac7d2670cde533d96c42837864623a0909/src/ueye_cam_driver.cpp#L1495)
```
IS_CM_BGR8_PACKED
IS_CM_BGRA8_PACKED
IS_CM_CBYCRY_PACKED
IS_CM_JPEG
```

Just the **additonal** feature to send in YCbCr is not supported by ueye_cam. 

So the camera in general works in ueye_cam. Furthermore the `xs_bayer_rggb.launch` file tries to use the color mode `IS_CM_SENSOR_RAW8` which is not supported by the camera (see list above or the IDS manual)

So, as the camera works (just not all features) I am closing this issue: ROS does not support YCbCr ([see here](http://docs.ros.org/en/noetic/api/sensor_msgs/html/namespacesensor__msgs_1_1image__encodings.html)) and using opencv or something else to decode YCbCr and encode it in BGR8 makes not much sense as the camera (or the IDS driver) is capable of providing a BGR8 image directly.

closed 2021-11-15 by nullket