anqixu/ueye_cam #95: Future of this repo

nullket, 2020-12-05:
**First of all**:
This is not a complaint. I just want to figure out where this project is going. 

**My rather long question for the maintainer:**
The project was almost dead and then @anqixu and @jmackay2 started to work on some things in the summer. But it is now stalled again for more than two months (again no complain, everybody is here in his/her freetime). 

I wonder how we could keep it going. I have forked (not public) the project longtime ago an integrated a lot of features (GPIO, triggers, a lot more IDS Parameters, etc.). After a lot of testing my adjustments, I wanted to push a couple of the changes upstream (here) and started pull request two months ago. But since then nothing happened, not even a discussion (and again not blaming just noting). I have even more features I would like to share and stuff to work on (e.g. some commands are deprecated in the newest ids software). 

A lot of stuff is outdated like copyright statements reaching back to 2013, the documentation could be improved, I have a couple more improvements of the code etc. Personally, I do not want to put effort in here if the maintainer do not actively review/accept/decline changes. I am happy to help from time to time - otherwise I would maybe do a fork but I not think it is helpful to split the development. What is your opinion?

anqixu, 2020-12-05:
Thank you @nullket for taking the effort, both to contribute to this project and for championing these thoughts.

Truth be told, I've abandoned hope of finding spare time to do active development on this project several years ago. I tried to squeeze time to maintain PRs over the summer, and lately, I have had a Jetson Nano + uEye combo on my desk for about a month (mainly because somehow the release builds are failing daily). But I've been struggling to find time while going through work & life transitions and typical sources of stress.

**To @nullket and other serious contributors with proven efforts**: if you wish to take over this project, I think it would be great for all involved: the community, you, and I. Please reach out to me privately ([my github handle] <at> cim <period> mcgill <dot> ca) and we can expedite the handover of ownership of the repo, release repo, wiki, and other matters.

nullket, 2020-12-07:
Thanks for the warm and honest words, I can totally relate (and as said above many are working on stuff like this in his/her freetime). I will send you an email right after that. 

But as you implicitly mentioned: I also think that just me helping out won't be enough. More contributors are needed. If somebody reads this and feels like he could help, please comment here! As in most community-driven opensource projects, helping can be from reviewing issues to just testing merge request to writing your own code.

jmackay2, 2020-12-09:
I don't know if there is too much need for active development at the moment. I think there are still some improvements like the PRs by @nullket , but I think the next step for this is probably just a noetic release and update to the IDS 4.94 software suite. Beyond that it is probably time for a rewrite into a ROS 2 component, which should probably be done in a separate repo.

nullket, 2020-12-09:
You are right. But  testing the PRs, creating new ones, fixing the 4.94. changes is development that needs to be done. I am willing to to do this (I need it anyways) but as this whole thing is about hardware it would always be beneficial that something will be tested by someone else. @jmackay2 I saw you creating PRs etc. can you maybe help and test PRs I will create to we can merge them without bothering @anqixu ?

jmackay2, 2020-12-10:
I can probably test PRs.

closed 2020-12-18 by jmackay2