anqixu/ueye_cam #96: Duplicating the rgb8.launch making issue

dt-developer1, 2020-12-06:
I have 3 Ueye cameras and i am using this package to publish streams to ros .  I duplicated  rgb8.launch for for each cmeras and i created 3 launch files. I changed node name , camera_name and camera_id for each camera. 

cam1.launch 
`<?xml version="1.0"?>
<launch>
  <node name="cam1" pkg="ueye_cam" type="cam1" required="true" />
  <arg name="nodelet_manager_name" default="nodelet_manager" />
  <arg name="camera_name" default="camera1" />

  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager_name)" args="manager" output="screen" />

  <node pkg="nodelet" type="nodelet" name="ueye_cam_nodelet"
        args="load ueye_cam/ueye_cam_nodelet $(arg nodelet_manager_name)">
    <param name="camera_name" type="str" value="$(arg camera_name)" /> 
    <param name="camera_topic" type="str" value="image_raw" />
    <param name="camera_id" type="int" value="0" /> `

cam2.lanch 

`<?xml version="1.0"?>
<launch>
  <node name="cam2" pkg="ueye_cam" type="cam2" required="true" />
  <arg name="nodelet_manager_name" default="nodelet_manager" />
  <arg name="camera_name" default="camera2" />

  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager_name)" args="manager" output="screen" />

  <node pkg="nodelet" type="nodelet" name="ueye_cam_nodelet"
        args="load ueye_cam/ueye_cam_nodelet $(arg nodelet_manager_name)">
    <param name="camera_name" type="str" value="$(arg camera_name)" /> 
    <param name="camera_topic" type="str" value="image_raw" />
    <param name="camera_id" type="int" value="3" /> `

cam3.launch 

`<?xml version="1.0"?>
<launch>
  <node name="cam3" pkg="ueye_cam" type="cam3" required="true" />
  <arg name="nodelet_manager_name" default="nodelet_manager" />
  <arg name="camera_name" default="camera3" />

  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager_name)" args="manager" output="screen" />

  <node pkg="nodelet" type="nodelet" name="ueye_cam_nodelet"
        args="load ueye_cam/ueye_cam_nodelet $(arg nodelet_manager_name)">
    <param name="camera_name" type="str" value="$(arg camera_name)" /> 
    <param name="camera_topic" type="str" value="image_raw" />
    <param name="camera_id" type="int" value="4" /> `

Now, when i am starting first cmera, it working and i am getting data in ros. when i am launching second or third camera after 1 camera , i am getting error in cam1.launch  like 

`[ INFO] [1607236696.973853566]: [camera1] set to free-run mode
[ERROR] [1607236711.938520154]: Cannot load nodelet /ueye_cam_nodelet for one exists with that name already
[ WARN] [1607236711.945882950]: Shutdown request received. Reason: [[/ueye_cam_nodelet] Reason: new node registered with same name]
[ WARN] [1607236711.947521305]: Shutdown request received.
[ WARN] [1607236711.947627881]: Reason given for shutdown: [[/nodelet_manager] Reason: new node registered with same name]
[FATAL] [1607236712.025064890]: Failed to unload nodelet '/ueye_cam_nodelet` from manager `nodelet_manager'
`

and in cam2.launch 

`process[nodelet_manager-2]: started with pid [29308]
process[ueye_cam_nodelet-3]: started with pid [29309]
[FATAL] [1607236711.938587608]: Failed to load nodelet '/ueye_cam_nodelet` of type `ueye_cam/ueye_cam_nodelet` to manager `nodelet_manager'
[ INFO] [1607236711.948214427]: Initializing nodelet with 12 worker threads.
[ERROR] [1607236712.024385110]: Failed to find nodelet with name '/ueye_cam_nodelet' to unload.
[ueye_cam_nodelet-3] process has died [pid 29309, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load ueye_cam/ueye_cam_nodelet nodelet_manager __name:=ueye_cam_nodelet __log:=/home/atc8110/.ros/log/f884f278-3784-11eb-877b-173023e66b60/ueye_cam_nodelet-3.log].
log file: /home/atc8110/.ros/log/f884f278-3784-11eb-877b-173023e66b60/ueye_cam_nodelet-3*.log
`
What will be the issue?

nullket, 2020-12-09:
1. The output is hard to read when using one single quote to mark them as code (inline code). Use three quotes ''' at the beginning (and three at the end) to create a code block preserving the line breaks. 

2. launchfiles should be .launch not .txt

3. This is due to the launch file trying to create a seconds nodelet manager (with the same name) etc (or similar issue). Anyway: to start multiple cameras you put the all in one launch file. To get an idea how this works look at the [master_slave_example](https://github.com/anqixu/ueye_cam/blob/master/launch/master_slaves_rgb8.launch) (but ignore the whole master -> slave part as it seems like you want three independent cameras)

Please close this issue once this is solved for you

nullket, 2020-12-12:
Closing as this is not a bug. It is the design of the driver. Nonetheless, you can still ask for help here.

closed 2020-12-12 by nullket