anqixu/ueye_cam #106: IS_INVALID_BUFFER_SIZE error when start X and Y other than 0

singhjeet1989, 2021-01-28:
Dear Team, 
I am working on a project which is based on IDS camera. I came across "IS_INVALID_BUFFER_SIZE" error while using some of configs. Those config files are generated from IDS cockpit SW. Most of the configuration works with ROS driver except some of them.
Here is exact error message: "**[ERROR] [1611661455.769880480]: Could not start free-run live video mode for [camera] (IS_INVALID_BUFFER_SIZE)**" 

I am also attaching the config files.
[config.zip](https://github.com/anqixu/ueye_cam/files/5886091/config.zip)

I have created my own c++ wrapper for IDS driver which can handle these config files. I have strong feeling that there is bug in ROS driver.

After investigation I found that this error arises only when start X and Y is other 0.

I also found the solution, I would like to check if this can be merged to official master branch.

anqixu, 2021-01-28:
Thank you very much for your feedback and contribution.

I think it makes sense. You added the section that checks if cam_aoi_.s32X > 0 || cam_aoi_.s32Y > 0, and if that's the case you get the max frame dimensions. It seems to match the `Start x absolute` and ...y... settings in your config files.

I haven't touched IDS for quite a while, and I don't have access to my camera for testing till April, since I'm relocating internationally. 

Could you make a pull request with your new code (it seems sensible and small enough). We can have others review + test your PR, before merging.

singhjeet1989, 2021-01-29:
@anqixu Thank you for response.
I created my own branch in order to create pull request but unfortunately I can't commit because of write access issue. Could you please grant me write access so that I can create pull request.

jmackay2, 2021-01-29:
@singhjeet1989 You should make a personal fork of the project and commit your changes there. Then you can make a pull request back to this project.

singhjeet1989, 2021-01-29:
Thanks @jmackay2, pull request is created.

closed 2021-06-29 by nullket