anqixu/ueye_cam #111: Multiple ueye_cams with criss-crossed streams

stonier, 2021-07-13:
Wondering if anyone here has this experience...

* Two ueye_cams with different device ids
* Two ueye_cam nodes with two instances `rqt_iamge_view` connecting to each node (so they should get into streaming mode)
* Both cameras connect, are in use, but both show the same image stream

This isn't a `ueye_cam` problem either. We observe the same effect with two launchers of `idscameramanager`.

@jmackay2 @anqixu

stonier, 2021-07-13:
FYI @ChrisFajardo-TRI

ChrisFajardo-TRI, 2021-07-13:
Here's a behavior we're reproducing

1. launch camera A node 
 camera A goes to streaming mode

2. launch camera B node
 camera B does not go to streaming mode (will timeout at is_CaptureVideo)

3. terminate and restart camera A node
camera A and camera B both go to streaming mode
however both camera A and camera B both stream camera A's images on both camera A and B topics

ChrisFajardo-TRI, 2021-07-14:
@stonier you can close this

found from idscameramanager that both cameras had multicast was enabled with same IP, so disabling multicast fixes things

stonier, 2021-07-14:
Might leave it open - would be nice to be able to *detect* the situation and warn from this driver and additionally have a means to configure it from here (instead of idscameramanager). 

@ChrisFajardo-TRI has pointed us at `is_Multicast` and `IS_AMC_CMD_SET_MC_ENABLED` in the IDS SDK as a possible means to do so.

jmackay2, 2021-07-15:
I haven't used multicast in my multi-camera setups in the past, so I haven't run into this issue before.