anqixu/ueye_cam #117: Galactic Release?

cco-seasony, 2021-08-04:
Hey! 
Any updates on a ROS2 Galactic release?

I am trying to build the foxy branch in galactic and basically debugging whatever comes up to make it work. Unfortunately we do not have the resources to do the full porting.

Thanks in advance!
Carlos

jmackay2, 2021-08-10:
I took a quick look at this and fixed a few of the bugs for Galactic (that should have been fixed in Foxy anyway). It now technically builds if you remove the `-Werror` flag in the `CMakeLists.txt`. 

Here are the two remaining warnings:

-  format security of `std::string` as a format string which is warned about in the [Galactic release page](https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#change-in-rclcpp-s-logging-macros)
- rclcpp::Duration deprecation

cco-seasony, 2021-08-10:
Hey! thanks for the reply.
I managed to fix the issues too but forgot to close the issue!
In any case, thanks a lot for taking the time.

closed 2021-08-10 by cco-seasony