anqixu/ueye_cam #126: multiple cameras

aristow1, 2022-11-11:
im trying to work with 4 IDS camera but im first testing with only 2. in the launch file i added a 2 second node and changed the camera_name param and the camera_id param. here is my launch file :

<?xml version="1.0"?>
<launch>
  <node name="check_ueye_api" pkg="ueye_cam" type="check_ueye_api" required="true" />
  <arg name="nodelet_manager_name" default="nodelet_manager" />
  <arg name="camera_name" default="" />

  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager_name)" args="manager" output="screen" />

  <node pkg="nodelet" type="nodelet" name="ueye_cam_nodelet"
        args="load ueye_cam/ueye_cam_nodelet $(arg nodelet_manager_name)">
    <param name="camera_name" type="str" value="camera1" /> 
    <param name="camera_topic" type="str" value="image_raw" />
    <param name="camera_id" type="int" value="1" /> 
    <param name="camera_intrinsics_file" type="string" value="file:$(find ueye_cam)/camera.yaml" /> 
    <param name="camera_parameters_file" type="string" value="file:$(find ueye_cam)/camera.ini" /> 
    <param name="ext_trigger_mode" type="bool" value="False" /> 

    

    <param name="color_mode" type="str" value="rgb8" /> 

    
    <param name="image_width" type="int" value="1600" />
    <param name="image_height" type="int" value="1200" />
    <param name="image_top" type="int" value="-1" /> 
    <param name="image_left" type="int" value="-1" /> 

    <param name="subsampling" type="int" value="1" /> 
    <param name="binning" type="int" value="1" /> 
    <param name="sensor_scaling" type="double" value="1.0" /> 

    <param name="auto_gain" type="bool" value="True" />
    <param name="master_gain" type="int" value="0" />
    <param name="red_gain" type="int" value="0" />
    <param name="green_gain" type="int" value="1" />
    <param name="blue_gain" type="int" value="16" />
    <param name="gain_boost" type="bool" value="False" />

    <param name="auto_exposure" type="bool" value="True" />
    <param name="exposure" type="int" value="33" /> 

    <param name="auto_white_balance" type="bool" value="True" />
    <param name="white_balance_red_offset" type="int" value="0" />
    <param name="white_balance_blue_offset" type="int" value="0" />

    <param name="flash_delay" type="int" value="0" /> 
    <param name="flash_duration" type="int" value="1000" /> 

    <param name="auto_frame_rate" type="bool" value="False" />
    <param name="frame_rate" type="double" value="30.0" />
    <param name="output_rate" type="double" value="0.0" /> 
    <param name="pixel_clock" type="int" value="25" />

    <param name="flip_upd" type="bool" value="False" />
    <param name="flip_lr"  type="bool" value="False" />
  </node>

                                        

     
  <node pkg="nodelet" type="nodelet" name="ueye_cam_nodelet2"
        args="load ueye_cam/ueye_cam_nodelet $(arg nodelet_manager_name)">
    <param name="camera_name" type="str" value="camera2" /> 
    <param name="camera_topic" type="str" value="image_raw" />
    <param name="camera_id" type="int" value="5" /> 
    <param name="camera_intrinsics_file" type="string" value="file:$(find ueye_cam)/camera.yaml" /> 
    <param name="camera_parameters_file" type="string" value="file:$(find ueye_cam)/camera.ini" /> 
    <param name="ext_trigger_mode" type="bool" value="False" /> 

    

    <param name="color_mode" type="str" value="rgb8" /> 

    
    <param name="image_width" type="int" value="1600" />
    <param name="image_height" type="int" value="1200" />
    <param name="image_top" type="int" value="-1" /> 
    <param name="image_left" type="int" value="-1" /> 

    <param name="subsampling" type="int" value="1" /> 
    <param name="binning" type="int" value="1" /> 
    <param name="sensor_scaling" type="double" value="1.0" /> 

    <param name="auto_gain" type="bool" value="True" />
    <param name="master_gain" type="int" value="0" />
    <param name="red_gain" type="int" value="0" />
    <param name="green_gain" type="int" value="1" />
    <param name="blue_gain" type="int" value="16" />
    <param name="gain_boost" type="bool" value="False" />

    <param name="auto_exposure" type="bool" value="True" />
    <param name="exposure" type="int" value="33" /> 

    <param name="auto_white_balance" type="bool" value="True" />
    <param name="white_balance_red_offset" type="int" value="0" />
    <param name="white_balance_blue_offset" type="int" value="0" />

    <param name="flash_delay" type="int" value="0" /> 
    <param name="flash_duration" type="int" value="1000" /> 

    <param name="auto_frame_rate" type="bool" value="False" />
    <param name="frame_rate" type="double" value="30.0" />
    <param name="output_rate" type="double" value="0.0" /> 
    <param name="pixel_clock" type="int" value="25" />

    <param name="flip_upd" type="bool" value="False" />
    <param name="flip_lr"  type="bool" value="False" />
  </node>

</launch>

and in the the terminal i get this when i roslaunch  the rgb8.launch file : 

[ INFO] [1668175192.168043450]: Initializing nodelet with 16 worker threads.
[ WARN] [1668175193.629248790]: Could not load [camera]'s sensor parameters file file:/home/kamag/simulation_ws/src/ueye_cam/camera.ini (IS_FILE_WRITE_OPEN_ERROR)
[ INFO] [1668175194.684726567]: UEye camera [camera] initialized on topic /camera/image_raw
Width:			1600
Height:			1200
Left Pos.:		-1
Top Pos.:		-1
Color Mode:		rgb8
Subsampling:		1
Binning:		1
Sensor Scaling:		1
Auto Gain:		1
Master Gain:		0
Red Gain:		0
Green Gain:		1
Blue Gain:		16
Gain Boost:		0
Software Gamma:		100
Auto Exposure:		1
Auto Exposure Reference:	128
Exposure (ms):		33
Auto White Balance:	1
WB Red Offset:		0
WB Blue Offset:		0
Flash Delay (us):	0
Flash Duration (us):	1000
Ext Trigger Mode:	0
Auto Frame Rate:	0
Frame Rate (Hz):	11.9325
Output Rate (Hz):	0
Pixel Clock (MHz):	25
GPIO1 Mode:	0
GPIO2 Mode:	0
Mirror Image Upside Down:	0
Mirror Image Left Right:	0

[ERROR] [1668175194.708981055]: Tried to advertise a service that is already advertised in this node [/camera/image_raw/compressedDepth/set_parameters]
[ERROR] [1668175194.713109108]: Tried to advertise a service that is already advertised in this node [/camera/image_raw/compressed/set_parameters]
[ERROR] [1668175194.716858114]: Tried to advertise a service that is already advertised in this node [/camera/image_raw/theora/set_parameters]
[ERROR] [1668175194.721680041]: Tried to advertise a service that is already advertised in this node [/camera/set_camera_info]

why is it showing /camera/image_raw although i changed the namespace to camera1 and camera2 and how can i fix this Problem.
Thank you.

aristow1, 2022-11-11:
the launch file was missing the first part

<launch>
  <node name="check_ueye_api" pkg="ueye_cam" type="check_ueye_api" required="true" />
  <arg name="nodelet_manager_name" default="nodelet_manager" />
  <arg name="camera_name" default="" />

  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager_name)" args="manager" output="screen" />

  <node pkg="nodelet" type="nodelet" name="ueye_cam_nodelet"
        args="load ueye_cam/ueye_cam_nodelet $(arg nodelet_manager_name)">
    <param name="camera_name" type="str" value="camera1" /> 
    <param name="camera_topic" type="str" value="image_raw" />
    <param name="camera_id" type="int" value="1" /> 
    <param name="camera_intrinsics_file" type="string" value="file:$(find ueye_cam)/camera.yaml" /> 
    <param name="camera_parameters_file" type="string" value="file:$(find ueye_cam)/camera.ini" /> 
    <param name="ext_trigger_mode" type="bool" value="False" /> 

    

    <param name="color_mode" type="str" value="rgb8" /> 

    
    <param name="image_width" type="int" value="1600" />
    <param name="image_height" type="int" value="1200" />
    <param name="image_top" type="int" value="-1" /> 
    <param name="image_left" type="int" value="-1" /> 

    <param name="subsampling" type="int" value="1" /> 
    <param name="binning" type="int" value="1" /> 
    <param name="sensor_scaling" type="double" value="1.0" /> 

    <param name="auto_gain" type="bool" value="True" />
    <param name="master_gain" type="int" value="0" />
    <param name="red_gain" type="int" value="0" />
    <param name="green_gain" type="int" value="1" />
    <param name="blue_gain" type="int" value="16" />
    <param name="gain_boost" type="bool" value="False" />

    <param name="auto_exposure" type="bool" value="True" />
    <param name="exposure" type="int" value="33" /> 

    <param name="auto_white_balance" type="bool" value="True" />
    <param name="white_balance_red_offset" type="int" value="0" />
    <param name="white_balance_blue_offset" type="int" value="0" />

    <param name="flash_delay" type="int" value="0" /> 
    <param name="flash_duration" type="int" value="1000" /> 

    <param name="auto_frame_rate" type="bool" value="False" />
    <param name="frame_rate" type="double" value="30.0" />
    <param name="output_rate" type="double" value="0.0" /> 
    <param name="pixel_clock" type="int" value="25" />

    <param name="flip_upd" type="bool" value="False" />
    <param name="flip_lr"  type="bool" value="False" />
  </node>

                                        

     
  <node pkg="nodelet" type="nodelet" name="ueye_cam_nodelet2"
        args="load ueye_cam/ueye_cam_nodelet $(arg nodelet_manager_name)">
    <param name="camera_name" type="str" value="camera2" /> 
    <param name="camera_topic" type="str" value="image_raw" />
    <param name="camera_id" type="int" value="5" /> 
    <param name="camera_intrinsics_file" type="string" value="file:$(find ueye_cam)/camera.yaml" /> 
    <param name="camera_parameters_file" type="string" value="file:$(find ueye_cam)/camera.ini" /> 
    <param name="ext_trigger_mode" type="bool" value="False" />

nullket, 2022-11-11:
Hello @aristow1 pleas format code/logs etc. with the standard markdown functions for that. This makes it much easier for other to read your post and be able to help you (I edited your post already for you).

- Remove the line `  <arg name="camera_name" default="" />`
- Are you sure that this launch file is REALLY the one that gets launched? Please create a complete new one, with a new unique name to make sure the correct one gets executed.
- Try another default launch file for two or more cameras like [this](https://github.com/anqixu/ueye_cam/blob/master/launch/master_slaves_rgb8.launch) (does not matter if you really want a master slave setup, just for testing this launch file should do the trick)
- Here is an untested minimal example which **should** work:
`test_two_cameras.launch`
```xml
<launch>
 <node name="check_ueye_api" pkg="ueye_cam" type="check_ueye_api" required="true" />
  <arg name="nodelet_manager_name" value="nodelet_manager" />
  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager_name)" args="manager" output="screen" />
<!--Stereo Camera left-->
  <node pkg="nodelet" type="nodelet" name="ueye_cam_nodelet_stereo_left"
        args="load ueye_cam/ueye_cam_nodelet $(arg nodelet_manager_name)">
    <param name="camera_name" type="str" value="stereo_left" /> <!-- == namespace for topics and services -->
    <param name="camera_topic" type="str" value="image_raw" />
    <param name="camera_id" type="int" value="1" />
    <param name="color_mode" type="str" value="rgb8" />                         <!-- valid options: 'rgb8', 'mono8', 'bayer_rggb8' -->
    <param name="image_width" type="int" value="1600" />
    <param name="image_height" type="int" value="1200" />
    <param name="image_top" type="int" value="-1" />                            <!-- -1: center -->
    <param name="image_left" type="int" value="-1" />                           <!-- -1: center -->

  </node>

  <!--Stereo Camera  right-->
  <node pkg="nodelet" type="nodelet" name="ueye_cam_nodelet_stereo_right"
        args="load ueye_cam/ueye_cam_nodelet $(arg nodelet_manager_name)">
    <param name="camera_name" type="str" value="stereo_right" /> <!-- == namespace for topics and services -->
    <param name="camera_topic" type="str" value="image_raw" />
    <param name="camera_id" type="int" value="5" />
    <param name="color_mode" type="str" value="rgb8" />                         <!-- valid options: 'rgb8', 'mono8', 'bayer_rggb8' -->
    <param name="image_width" type="int" value="1600" />
    <param name="image_height" type="int" value="1200" />
    <param name="image_top" type="int" value="-1" />                            <!-- -1: center -->
    <param name="image_left" type="int" value="-1" />                           <!-- -1: center -->

  </node>

</launch>

```

aristow1, 2022-11-14:
thank you this helped me ! and then i just can add as much cameras as i want to the launch file with their respective names and ID right ?

nullket, 2022-11-14:
Yes, you need to set the name and ID.

please close the issue if it is resolved

closed 2022-11-14 by aristow1