anqixu/ueye_cam #127: can't load .ini file

aristow1, 2022-11-14:
i keep getting this warning  and i don't know what it means : "INVALID_CAMERA_TYPE" 

`[ WARN] [1668423712.167040860]: Could not load [ids_middle]'s sensor parameters file /home/aristow/simulation_ws/src/ueye_cam/ids_middle.ini (IS_INVALID_CAMERA_TYPE)`

this is my .ini file : 

```
# Camera intrinsics

[image]

width
1600

height
1200

[narrow_stereo]

camera matrix
1355.21561 0.00000 792.83327 
0.00000 1349.11384 584.57143 
0.00000 0.00000 1.00000 

distortion
-0.42508 0.19559 0.00350 0.00026 0.00000 

rectification
1.00000 0.00000 0.00000 
0.00000 1.00000 0.00000 
0.00000 0.00000 1.00000 

projection
1134.58801 0.00000 790.87401 0.00000 
0.00000 1227.31055 585.41224 0.00000 
0.00000 0.00000 1.00000 0.00000 
```
any idea How to solve this or what this is even is ? the camera type is narrow_stereo i think, no ?

Philip-AH, 2022-11-23:
You need to specify the right sensor type. For example:

[Sensor]
Sensor=UI527xCP-C

nullket, 2022-11-23:
In addition to @Philip-AH perfect answer I would like to point out that everything that starts with `IS_*` is usually something from the underlying ueye sdk. I recommend looking up those uppercase `IS_*` messages in the sdk manual to figure out what the issue might be. 

Unfortunately, you need to register yourself on the IDS Website but it is completely free. Your mentioned error is explained here: https://en.ids-imaging.com/manuals/ids-software-suite/ueye-manual/4.95.1/en/is_camerastatus.html  

Also if you really want to use the .ini feature, I would recommend to create it with the ids camera manager/suite and only fine tune it using a text editor. This also prevents stuff like AOI-Crops or image sizes the sensor does not allow (all mentioned in the SDK documentation) as the IDS GUI is so clever to only allow setting values that are possible. 

Nonetheless, if you do not have a fully perfect ini file yet, I would recommend to overthink your approach and might simply set all parameters in the launch file and have only the tradtional ros yaml file for the camera intrinsics like this:
```xml
<param name="camera_intrinsics_file" type="string" value="$(find package_name_where_you_stored_the_file)/params/result_of_camera_calibration.yaml"`) />
```

closed 2022-11-24 by aristow1