anqixu/ueye_cam #130: Failed to build ueye_cam in ROS2

simk0024, 2023-03-22:
I encounter error while building the `ueye_cam` package in ROS2

My environment: 
- OS: Ubuntu 22.04
- ROS: ROS2 Humble
- IDS uEye Software Suite: 4.94.2

Error message:
```
--- stderr: ueye_cam
/home/sg-atcoe-23761/ws_scannerbot/src/driver/ueye_cam/src/driver/driver.cpp: In static member function ‘static const string ueye_cam::Driver::colormode2name(INT)’:
/home/sg-atcoe-23761/ws_scannerbot/src/driver/ueye_cam/src/driver/driver.cpp:1873:43: error: loop variable ‘value’ of type ‘const std::pair<std::__cxx11::basic_string<char>, int>&’ binds to a temporary constructed from type ‘const std::pair<const std::__cxx11::basic_string<char>, int>’ [-Werror=range-loop-construct]
 1873 |   for (const std::pair<std::string, INT>& value: COLOR_DICTIONARY) {
      |                                           ^~~~~
/home/sg-atcoe-23761/ws_scannerbot/src/driver/ueye_cam/src/driver/driver.cpp:1873:43: note: use non-reference type ‘const std::pair<std::__cxx11::basic_string<char>, int>’ to make the copy explicit or ‘const std::pair<const std::__cxx11::basic_string<char>, int>&’ to prevent copying
cc1plus: all warnings being treated as errors
gmake[2]: *** [src/driver/CMakeFiles/ueye_cam_driver.dir/build.make:76: src/driver/CMakeFiles/ueye_cam_driver.dir/driver.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:240: src/driver/CMakeFiles/ueye_cam_driver.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ueye_cam [0.49s, exited with code 2]
```

Any fix or advice?

Thank you

simk0024, 2023-03-22:
Solution

in `src/driver/driver.cpp`, line 1872, remove the `&`.

```
const std::string Driver::colormode2name(INT mode) {
  for (const std::pair<std::string, INT> value: COLOR_DICTIONARY) {
    if (value.second == mode) {
      return value.first;
    }
  }
  return std::string();
}
```

federkamm, 2023-03-22:
Or add a `const` to make the line read:

```c++
for (const std::pair<const std::string, INT>& value: COLOR_DICTIONARY) // ...
```

jmackay2, 2023-03-29:
This has been fixed in #129 .